Blockstream/satellite #4: segfault under ubuntu 16.04

honzik666, 2017-08-16:
The modules seem to crash, with a stock GNU radio in Ubuntu 16.04. I am assuming this is an incompatibility between the OOT modules and the distribution gnu radio:

```
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1250000.002070 Hz
[R82XX] PLL not locked!
Using Volk machine: avx2_64_mmx_orc
Traceback (most recent call last):
  File "./rx_gui.py", line 1029, in <module>
    main()
  File "./rx_gui.py", line 1017, in main
    tb = top_block_cls(freq=options.freq, gain=options.gain, loopbw=options.loopbw, loopbw_0=options.loopbw_0, fllbw=options.fllbw)
  File "./rx_gui.py", line 196, in __init__
    self.qtgui_time_sink_x_2.enable_axis_labels(True)
AttributeError: 'time_sink_c_sptr' object has no attribute 'enable_axis_labels'
Segmentation fault (core dumped)

```

blockstreamsatellite, 2017-08-16:
What version of GNUradio are you using?

honzik666, 2017-08-18:
this was with 3.7.9, I am switching to pybombs setup now as there is no newer version of gnuradio for ubuntu 16.04 lts.

weex, 2017-09-02:
@honzik666 Did you have any luck with the PyBOMBS setup? About to blow away my Ubuntu install to go with grubles Fedora setup.

honzik666, 2017-09-04:
Yes, I was able to install most of it and also the gr-framers package. The other part - I had to compile and install but that worked fine. I believe, once it has its own recipe, the installation can also run fully through pybombs

aecs18, 2018-07-30:
Good afternoon everyone.I write here to see if any of you know that installation errors I may have had to use the r820t2 in gnu radio companion.

I'm trying to make an FM receiver and other analog TV receiver but when running the program does not receive any type of signal and the view errors, I have: devices not supported. I believe that this is some driver to be installed but I don't know where or how.

The steps that I have followed for installation (Ubuntu 16.04 LTS) have been:

1. Install gnuradio.
2. Install Xenial Xerius specifications (it is supposed to do I need for my version of Ubuntu).
3. Install specifications of gr-osmosdr to enable some features of the chip.

The Stick comes with a CD but i do not know how to install it since, according to the specifications only works for WXP, 2000, Vista, 7 and 8; and it is something that it seems strange to me since I have seen videos where they use this same model of stick to carry out projects in Gnuradio on Ubuntu.

Please, if any of you know that i need to install It would serve as much help as I am a novice programmer and i need to resolve this problem for my project.

closed 2019-04-17 by blockstreamsatellite