Blockstream/satellite #6: Please confirm frequency for Galaxy 18

aospan, 2017-08-20:
[image]

I see signal on not expected frequency. I have LNB with 10750 heterodyne LO freq and dish pointed to galaxy 18. According to info from https://blockstream.com/satellite/satellite/ I should see signal at 12178.43 - 10750 = 1428,43 MHz but as you can see on my screenshot above there is some signal on 1429 and no signal on expected 1428,43. And I see "##### Frame synchronization acquired #####", so at least 'frame_sync_fast_impl' is happy ? But there is no any data goes to '/tmp/async_rx'.
Can you please confirm that my observations is correct ?

P.S.
I can receive dvb-s streams from this satellite on declared frequency (for example, 12033 H), so lnb&dish should work fine.

blockstreamsatellite, 2017-08-20:
That all looks correct and if you have frame lock, you should see data output on /tmp/async_rx. 

Could you send a screenshot of the Costas Sym Out?

aospan, 2017-08-21:
Now it works, thanks ! Seems like I didn't wait enough time ... 
Now I see UDPMessage's comes from /tmp/async_rx and have meaningful data (MSG_TYPE_BLOCK_HEADER and MSG_TYPE_BLOCK_CONTENTS):
```
aospan: msg_type=4 hash_prefix=0x72a99283 chunks_sent=1744 chunk_id=1241
aospan: msg_type=4 hash_prefix=0x72a99283 chunks_sent=1744 chunk_id=1242
aospan: msg_type=4 hash_prefix=0x72a99283 chunks_sent=1744 chunk_id=1243
```

here is a screenshot of the Costas Sym Out. Looks good for QPSK:

[image]

Continue investigation ... 
Thanks for amazing stuff ! :)

closed 2017-08-21 by aospan