Blockstream/satellite #8: Received blocks questions

aospan, 2017-08-21:
Hello !

I'm checking ~/.bitcoin/debug.log and see that received blocks are processed but with error messages:
```
2017-08-21 18:59:53 ERROR: AcceptBlockHeader: prev block not found
2017-08-21 18:59:53 ERROR: ProcessNewBlock: AcceptBlock FAILED
2017-08-21 18:59:53 UDP: Failed to decode block 0000000000000000012d5fb1780ae6cfd81c732617b1974457546cc4a9f582b1
```

Should I fully sync my node over internet before switching to satellite updates ?

This block hash is for 'BITCOIN BLOCK 481401' which has arrived about 30 hours ago (source: https://www.blocktrail.com/BTC/block/481401). Is this expected delay between current blockchain state and broadcasted over satellite ?

blockstreamsatellite, 2017-08-21:
It is best to start with a fully synced node.  We also broadcast the previous 24 hours of blocks (or approximately 24 hours of blocks, its actually the last 144 blocks which is typically 24 hours).  This is why you're seeing an older block.  

When a new block arrives, we stop sending the old data, broadcast the new block, and then resume sending the old blocks.

aospan, 2017-08-21:
great, thanks !

closed 2017-08-21 by aospan