Blockstream/satellite #9: Suggestion; a hardware proven to work list

MatzFan, 2017-08-24:
Would like to set up a sat node, but hardware is proving a stumbling block (no pun intended :smile:). [grubles](https://medium.com/@notgrubles/building-your-own-bitcoin-satellite-node-6061d3c93e7) has a great guide, but Amazon (& now ebay) have already sold out of the recommended Maverick MK1-PLL LNB. A list of LNB's (& compatible power injectors) proven to work may be useful to sat node noobs, like me.

blockstreamsatellite, 2017-08-24:
We are working with a vendor to provide greater availability of the components.  Also, we have a software update coming soon that should eliminate the need for a high stability LNB. Once we release this, any LNB will work as long as the frequency range is correct.

MatzFan, 2017-08-24:
Perfect, thanks

closed 2017-08-24 by blockstreamsatellite