Blockstream/satellite #13: Please provide either the brand or vendor who provides the hardware, thanks.

yrrv1980, 2017-09-18:


royhodge830, 2019-06-24:
required hardware - software defined radio (SDR) interface, satellite dish/antenna, low-noise block downconverter (LNB), power supply, mounting hardware, cables and connectors.

I believe Blockstream has already started producing custom hardware units of it's own, I as a hobbyist have built one or two with some friends and am designing a low cost DIY option as well. More info can be found here eventually, once I update it with newer Prototype design images & information. https://unibitlabs.github.io/blockteam.website/  Not sure where Blockstream's unit information can be found, but here's an image I found <img>https://pbs.twimg.com/media/DdUiN-SXUAUp69D.jpg</img>

blockstreamsatellite, 2020-05-06:
Hi @royhodge830 . FYI, there are kits available now at our [store](https://store.blockstream.com/product-category/satellite_kits/) and more information in the [hardware guide](https://github.com/Blockstream/satellite/blob/master/doc/hardware.md). The parts are for Blockstream Satellite 2.0, the new version released on Monday (May 4th 2020).

We will close this issue. Feel free to reopen if you have more questions.

closed 2020-05-06 by blockstreamsatellite