Blockstream/satellite #15: Signal reception question. Peak moves from left to right

steven-bloom, 2017-11-01:
Hello everyone! I have just Implemented Blockstream-receiver with RTL-SDR. No errors occurred during installation. However, i do not see peak than remains in a same place, instead of that  i receive signal that continuously moves from left to right (see screenshot)
What causes that behavior? 
In order to connect with Telstar 11N I use following LNBs: 
Kathrein UAS 584 (QuadLNB) and UAS 572 (Twin-LNB)
Key: 120 cm 
Azimut: 237,2° Elevation: 15,3° Polarisation: 32° (Skew-Angle)
Polarisation and Frequency band: Vertical Low
What settings can be tweaked  to get better and stable results? 
Thanks in advance. Any help would be greatly appreciated
[image]

blockstreamsatellite, 2017-11-01:
I suspect the signal is not strong enough to keep a lock.  Could you send a screen shot of the FLL IN and FLL out pages?

steven-bloom, 2017-11-02:
Hello, Blockstreamsatellite. :)
Thank you very mutch for your quick responce. 
These are Screenschots from FFL 
[image]
[image]
May there be some tips for setting up? I am trying to implement that in Germany.
Thanks in advance

blockstreamsatellite, 2017-11-02:
Yep, it is as I suspected, the signal looks a little too weak to maintain a lock.  Can you try adjusting the polarization (rotation) of your LNB to get a better peak on the FLL page?  Also, might try making adjustments to the pointing to see if you can get it any better.

Also, can you set the average value (in the corner on the FLL pages of the gui) to 20 and send some more screen shots?

What size satellite dish are you using?

steven-bloom, 2017-11-10:
Hi, everyone. I finally got a stable and good signal 
Now the question arises of how to know that the blocks are synchronized. I do not see any effect when running
bitcoind -fecreaddevice=/tmp/async_rx

I compiled Bitcoin engine (having first established Berkeley 4.8 and dependencies)  from this source https://github.com/bitcoinfibre/bitcoinfibre

./autogen.sh
/configure.sh
make 
sudo make install

I use Ubuntu 16.04 x64 LTS

Where can i find the debug.log (the file were referring to in this tutorial https://hackernoon.com/building-your-own-bitcoin-satellite-node-part-3-dish-alignment-1306b4c21326)

[image]

Thanks in advance

instagibbs, 2017-11-10:
> bitcoinfibre/bitcoinfibre
> satellite-node-part-3-dish-alignment-1306b4c21326)

On Ubuntu it's normally in ~/.bitcoin/debug.log

steven-bloom, 2017-11-11:
Hi, Gregory
Thank you very much
I was able to open this file and saw the following content. It does not look like the program is working correctly on my machine. Tell me please, what could be the problem
Any help would be greatly appreciated
[image]

blockstreamsatellite, 2017-11-14:
Are you starting with an already synced node?  I do not see any errors here.  Can you also run with the -debug flag.

Thank You

steven-bloom, 2017-11-29:
Hi to all. 
I have synchronized blockchain until last 24 hours. Then i ran 
bitcoind -fecreaddevice=/tmp/async_rx -printtoconsole -debug
Nothing appeared. 
[image]

Could you please tell me how one can view the contents of the file async_rx ? 
I have a suspicion that the file does not receive data.

I use external HDD as a Core-Source, Can this cause a problem?
Thanks in advance

steven-bloom, 2017-12-07:
everything has worked thank you!

closed 2017-12-07 by steven-bloom