Blockstream/satellite #18: Is signal transmitting now ?

aospan, 2018-01-10:
Hello,
I'm on Galaxy 18 (123W). Signal power is "flat", no constellation. Is signal transmitting now ?
DVB transponders works fine on the same time, so dish and lnb is ok.
Thanks !

P.S.
Is it ok to ask such questions here on github ? Please, let me know if it's better to use another communication channel (email, etc).

blockstreamsatellite, 2018-01-10:
We are looking into this now and will update soon.

Thanks!

blockstreamsatellite, 2018-01-10:
Everything appears to be operational.  Can you double check on your end?

aospan, 2018-01-10:
Now it works ! thanks !

aospan, 2018-02-19:
Again no signal ;( please check. Thanks !

blockstreamsatellite, 2018-02-20:
We actually had to move the frequency at the last minute. We have a public announcement coming shortly.  The new frequency is 12022.85 Mhz

aospan, 2018-02-21:
Thanks for update ! 
Should it work now on new frequency ?

blockstreamsatellite, 2018-02-21:
Yes, if you switch to the new frequency, you should be good to go!

aospan, 2018-02-21:
Yes, it works ! But not so good as before ... constellations looks very bad:

[image]

and stream is recovered with errors (as expected).
I didn't changed equipment and dish alignment. 
Is something changed on your side (other than frequency) ?

Thanks !

closed 2019-04-17 by blockstreamsatellite