Blockstream/satellite #19: No rule to make target 'swig/framers_swig.py', needed by 'swig/framers_swig.pyc'

grubles, 2018-07-30:
On a fresh Fedora 26 install, with the required dependencies installed to build, `./install_gr_framers.sh` fails with the following: 

```
Scanning dependencies of target gnuradio-framers
[  4%] Building CXX object lib/CMakeFiles/gnuradio-framers.dir/gr_hdlc_framer_b_impl.cc.o
[  8%] Building CXX object lib/CMakeFiles/gnuradio-framers.dir/gr_hdlc_deframer_b_impl.cc.o
[ 13%] Building CXX object lib/CMakeFiles/gnuradio-framers.dir/hdlc_framer.cc.o
[ 17%] Building CXX object lib/CMakeFiles/gnuradio-framers.dir/hdlc_deframer.cc.o
[ 21%] Building CXX object lib/CMakeFiles/gnuradio-framers.dir/gr_hdlc_framer_2_channel_b_impl.cc.o
[ 26%] Linking CXX shared library libgnuradio-framers.so
[ 26%] Built target gnuradio-framers
Scanning dependencies of target test-framers
[ 30%] Building CXX object lib/CMakeFiles/test-framers.dir/test_framers.cc.o
[ 34%] Building CXX object lib/CMakeFiles/test-framers.dir/qa_framers.cc.o
[ 39%] Linking CXX executable test-framers
[ 39%] Built target test-framers
Scanning dependencies of target framers_swig_swig_doc
[ 39%] Built target framers_swig_swig_doc
Scanning dependencies of target _framers_swig_swig_tag
[ 43%] Building CXX object swig/CMakeFiles/_framers_swig_swig_tag.dir/_framers_swig_swig_tag.cpp.o
[ 47%] Linking CXX executable _framers_swig_swig_tag
[ 47%] Built target _framers_swig_swig_tag
Scanning dependencies of target pygen_swig_ff46b
make[2]: *** No rule to make target 'swig/framers_swig.py', needed by 'swig/framers_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:273: swig/CMakeFiles/pygen_swig_ff46b.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
[ 26%] Built target gnuradio-framers
[ 39%] Built target test-framers
[ 39%] Built target framers_swig_swig_doc
[ 47%] Built target _framers_swig_swig_tag
make[2]: *** No rule to make target 'swig/framers_swig.py', needed by 'swig/framers_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:273: swig/CMakeFiles/pygen_swig_ff46b.dir/all] Error 2
```

grubles, 2018-07-30:
What is especially strange is that I can spin up a Fedora 26 VM and build everything successfully, but the box I'm installing Satellite on fails to build.

blockstreamsatellite, 2019-03-12:
@grubles we are closing some old issues. I suppose this was resolved? Are you still running on Fedora 26 today? OK to close the issue?

grubles, 2019-03-16:
I never figured out the issue, but since then there's been some non-backwards-compatible updates so it's probably best to close this. Fedora 26 is EOL as well.

closed 2019-03-16 by grubles