Blockstream/satellite #26: Failure to read API FIFO causes fibre data to stop

gmaxwell, 2019-01-16:
It looked like signal reception stopped to me:

In the fibre logs I saw 2019-01-16 19:11:49.224130 UDP[35]: Average speed 0.0894 Mbit/sec
In the modem logs I saw [2019-01-16 19:12:24] Rx DataI Blocks: 31.28 MBIUser API: 53.97 KB

And then no UDPstat or "Rx Data" for a half hour.

Based on the API number being kinda close to the default linux fifo maximum,  I tried 

$ cat /tmp/blocksat/api > /dev/null

And then suddenly the modem reported,

[2019-01-16 19:12:24] Rx DataI Blocks: 31.28 MBIUser API: 53.97 KB
[2019-01-16 20:30:36] Rx DataI Blocks: 31.34 MBIUser API: 58.52 KB
[2019-01-16 20:30:46] Rx DataI Blocks: 83.6 MBIUser API: 186.81 KB

and Fibre reported: 2019-01-16 20:31:36.872182 UDP[35]: Average speed 6.9333 Mbit/sec

And it caught back up.

Writes to fibre or the API should not block each other.

blockstreamsatellite, 2019-01-16:
@gmaxwell we will investigate how to improve this. In the meantime, if you are not using the API, can you try launching the receiver with `--no-api` command-line argument?

gmaxwell, 2019-01-16:
Restarted with that, as an aside-- --no-api isn't in the help output.

blockstreamsatellite, 2019-01-16:
@gmaxwell Thanks for letting us know. Which application are you using? Would you mind pasting the help output?

gmaxwell, 2019-01-16:
darn it, sorry, I was scrolled back to the output from the old version. It's there.

blockstreamsatellite, 2019-01-16:
@gmaxwell No problem. Thanks for pointing out the issue. We will push a fix to avoid having to use `--no-api` ASAP.

blockstreamsatellite, 2019-01-17:
@gmaxwell Would you like to try PR #27 ? Please note that it involves associated changes to the gr-blocksat repo. Hence, first you need to update your gr-blocksat submodule to the new commit that it points to (which has been udpated in the PR). After that, the usual:

```
make blocksat
sudo make install-blocksat
```

Then build satellite normally.

Let me know how it goes, if you find a chance. I could test what happens when the `bitcoinfibre` pipe is not being read, in which case there should be some overflow messages in the console. However, I haven't fully tested when the API pipe overflows yet.

commit ccb4451a (2019-01-21): Add option to disable output of Bitcoin FIBRE data

- Add parameter that allows disabling reception and the corresponding output of
  blocks data into the Bitcoin FIBRE named pipe.

- Update all instances of the Protocol Sink block from the gr-blocksat library
  and connect the value of the referred parameter (that disables blocks output)
  to the "disable_blocks" option of the sink blocks.

- Update Makefile to make the "--no-blocks" command-line argument a boolean,
  just like the previous "--no-api" argument.

- Update the submodule pointer to the latest gr-blocksat commit, which includes
  [... cut]
PR #27 (2019-01-21): Fix protocol sink blocking issue
- Add parameter that allows disabling reception and the corresponding output of
  blocks data into the Bitcoin FIBRE named pipe.

- Update all instances of the Protocol Sink block from the gr-blocksat library
  and connect the value of the referred parameter (that disables blocks output)
  to the "disable_blocks" option of the sink blocks.

- Update Makefile to make the "--no-blocks" command-line argument a boolean,
  just like the previous "--no-api" argument.

- Update the submodule pointer to the latest gr-blocksat commit, which includes
  the associated cha [... cut]
commit e0cd9d34 (2019-01-21): Add option to disable output of Bitcoin FIBRE data

- Add parameter that allows disabling reception and the corresponding output of
  blocks data into the Bitcoin FIBRE named pipe.

- Update all instances of the Protocol Sink block from the gr-blocksat library
  and connect the value of the referred parameter (that disables blocks output)
  to the "disable_blocks" option of the sink blocks.

- Update Makefile to make the "--no-blocks" command-line argument a boolean,
  just like the previous "--no-api" argument.

- Update the submodule pointer to the latest gr-blocksat commit, which includes
  [... cut]
closed 2019-01-21 by blockstreamsatellite