Blockstream/satellite #28: Outside coverage map

dxtrader, 2019-01-20:
Awesome project, was starting to see what parts I already had lying around and realized I'm located outside the coverage map.  Is it still possible to connect or will signal be too weak? Just go with a bigger dish?

blockstreamsatellite, 2019-04-17:
@dxtrader we are closing some old issues and have realized this issue was never responded. Apologies.

What is your location?

dxtrader, 2019-12-25:
Location is UAE and Saudi Arabia

blockstreamsatellite, 2020-05-06:
@dxtrader apologies for not replying earlier. We don't have coverage for your location yet, but it is still in the roadmap. You can check the [coverage map in our website](https://blockstream.com/satellite/#satellite_network-coverage).

closed 2020-05-06 by blockstreamsatellite