Blockstream/satellite #29: Which bitcoinfibre version is needed to synchronize the Bitcoin Blockchain with current Blockstream Satellite data?

VDelport, 2019-01-24:
I received Blockchain data via Blockstream Satellite today (24.01.19) with the receiver version rx_gui.py before the update in December 2018. The data can be received (100 MB in 2 hours), but bitcoinfibre version v0.15.0.0-29587c411 do not care about the data. Has bitcoinfibre been changed as well?

blockstreamsatellite, 2019-01-24:
Yes, bitcoinfibre should be from the master branch there in [bitcoinfibre’s repo](https://github.com/bitcoinfibre/bitcoinfibre).

See the note about the Blockstream Satellite Jan 16 update in our [readme](https://github.com/Blockstream/satellite/blob/master/README.md).

blockstreamsatellite, 2019-03-12:
@VDelport 

Do you still have an issue with this? Are you passing the correct named pipe (at `/tmp/blocksat/bitcoinfibre`) to bitcoin fibre? See the info in the wiki [Running-the-System#run-bitcoin-fibre](https://github.com/Blockstream/satellite/wiki/Running-the-System#run-bitcoin-fibre).

closed 2019-03-12 by VDelport