Blockstream/satellite #31: Issue running blocksat-rx on a Raspberry

philipp634, 2019-02-24:
Compiling and installing the binaries works fine. Running `blocksat-rx` or `blocksat-rx-lower` gives the following error

```
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
*** Error in `python': corrupted double-linked list: 0x01328cc8 ***
/usr/local/bin/blocksat-rx-lower: line 2: 20414 Aborted                 /usr/bin/env python /usr/local/lib/blocksat-rx/blocksat_rx_lower.pyc "$@"
```
Linux is a Raspbian Stretch Lite `Linux 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`

Is it possible to run blocksat on a Raspberry?

blockstreamsatellite, 2019-02-24:
Hello @philipp634 .

This is something that we've seen before. It is not related to blocksat specifically. It seems to be either an issue with hardware (CPU/memory) limitations of Raspberry Pi for running the receiver or related to gr-osmosdr. See for example [this thread](https://www.reddit.com/r/GNURadio/comments/8bjcsw/error_when_running_gnu_radio_on_raspberry_pi/). 

As of today, we are not aware of any functional blocksat receiver setups that use Raspberry Pi. If you get it to work, please let us know.

philipp634, 2019-02-24:
Hello, thanks for the quick answer. 

It seems like I have to look into a more powerful single-board computer like ODROID or similar in order to have a slim setup on the rooftop and getting the data in via WI-FI.

closed 2019-02-24 by philipp634

blockstreamsatellite, 2019-04-19:
Hello @philipp634 ,

Just so you know, the latest release (`v1.4.0`) is confirmed to work on Raspberry Pi 3 B+. Have a look at the [instructions in the README](https://github.com/Blockstream/satellite#running-on-raspberry-pi) page if interested.

BR