Blockstream/satellite #32: No signal with blocksat_rx while getting one with gqrx

darosior, 2019-03-02:
We have tried using the satellite (from France) for a while. We compiled it on Debian 10, Ubuntu 18, Ubuntu 16, Fedora 28 and Fedora 29, on three different computers. We got a great signal with all systems using `gqrx-sdr`, but never with `blocksat_rx` (I even tried to use older commits).
Here is a Dockerfile that we wanted to add it via a PR but which sums up what we did.
```
FROM fedora:28

RUN dnf update -y && dnf install -y gnuradio gnuradio-devel gr-osmosdr dnf-plugins-core git make cmake swig pkg-config doxygen graphviz gcc-c++ cppunit-devel

RUN dnf copr -y enable blockstream/satellite && dnf install -y satellite 

RUN git clone https://github.com/blockstream/satellite /opt/satellite

RUN cd /opt/satellite && make framers && make install-framers && make blocksat && make install-blocksat && export PYTHONPATH=/usr/local/lib64/python2.7/site-packages:$PYTHONPATH && export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH && export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH && make && make install
```
Here is the output : 
```
[root@efc39f74247e /]# blocksat-rx -f 1754020000 -g 49.6
linux; GNU C++ version 8.0.1 20180207 (Red Hat 8.0.1-0.12); Boost_106600; UHD_003.010.003.000-0-unknown

gr-osmosdr v0.1.x (0.1.5git) gnuradio 3.7.13.4
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 937500.020955 Hz
[R82XX] PLL not locked!
[2019-03-02 18:48:23] SNR [=                                       ] 0.0000 dB
Press Enter to quit: [2019-03-02 18:48:24] SNR [=======                                 ] 3.0778 dB
[2019-03-02 18:48:25] SNR [=======                                 ] 3.0936 dB
[2019-03-02 18:48:26] SNR [=======                                 ] 3.0714 dB
[2019-03-02 18:48:27] SNR [=======                                 ] 3.0835 dB
[2019-03-02 18:48:28] SNR [=======                                 ] 3.0786 dB
[2019-03-02 18:48:29] SNR [=======                                 ] 3.0688 dB
[2019-03-02 18:48:30] SNR [=======                                 ] 3.0698 dB
[2019-03-02 18:48:31] SNR [=======                                 ] 3.0876 dB
--------------------------------------------------------------------------------
[2019-03-02 18:48:31] Frame Timing => SEARCHING	 Timing Indicator: STRONG
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 18:48:31] Bit Error Rate: 0.00E+00
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 18:48:31] Carrier Frequency Offset: -1.8307 kHz 
--------------------------------------------------------------------------------
[2019-03-02 18:48:32] SNR [=======                                 ] 3.0509 dB
[2019-03-02 18:48:33] SNR [=======                                 ] 3.0716 dB
[2019-03-02 18:48:34] SNR [=======                                 ] 3.0711 dB
[2019-03-02 18:48:35] SNR [=======                                 ] 3.0779 dB
[2019-03-02 18:48:36] SNR [=======                                 ] 3.0868 dB
[2019-03-02 18:48:37] SNR [=======                                 ] 3.0864 dB
[2019-03-02 18:48:38] SNR [=======                                 ] 3.0670 dB
[2019-03-02 18:48:39] SNR [=======                                 ] 3.0754 dB
[2019-03-02 18:48:40] SNR [=======                                 ] 3.0944 dB
[2019-03-02 18:48:41] SNR [=======                                 ] 3.0757 dB
--------------------------------------------------------------------------------
[2019-03-02 18:48:41] Frame Timing => SEARCHING	 Timing Indicator: STRONG
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 18:48:41] Bit Error Rate: 0.00E+00
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 18:48:41] Carrier Frequency Offset: -2.5807 kHz 
--------------------------------------------------------------------------------
[2019-03-02 18:48:42] SNR [=======                                 ] 3.0663 dB
[2019-03-02 18:48:43] SNR [=======                                 ] 3.0500 dB
^CTraceback (most recent call last):
  File "build/blocksat_rx.py", line 645, in <module>
    main()
  File "build/blocksat_rx.py", line 637, in main
    raw_input('Press Enter to quit: ')
KeyboardInterrupt
[root@efc39f74247e /]# 
```

Have you a suggestion to get `blocksat_rx` working ?

NB : We also checked other related issues such as #30, but we have obviously already tried to disconnect, reconnect the SDR.

darosior, 2019-03-02:
Without the docker image, on Debian 10 and running `build/blocksat_rx_gui.py -f 1754020000 -g 49.6` we get this output 
```
[2019-03-02 19:53:46] Frame Timing => SEARCHING	 Timing Indicator: STRONG
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 19:53:46] Bit Error Rate: 3.18E-09
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 19:53:46] Carrier Frequency Offset: 29.2831 kHz 
--------------------------------------------------------------------------------
[2019-03-02 19:53:47] SNR [=======                                 ] 3.0874 dB
[2019-03-02 19:53:48] SNR [=======                                 ] 3.0551 dB
[2019-03-02 19:53:49] SNR [=======                                 ] 2.9689 dB
[2019-03-02 19:53:50] SNR [=======                                 ] 3.0259 dB
[2019-03-02 19:53:51] SNR [=======                                 ] 2.9849 dB
[2019-03-02 19:53:52] SNR [=======                                 ] 3.1844 dB
[2019-03-02 19:53:53] SNR [=======                                 ] 3.1422 dB
[2019-03-02 19:53:54] SNR [=======                                 ] 2.9565 dB
[2019-03-02 19:53:55] SNR [=======                                 ] 3.0502 dB
[2019-03-02 19:53:56] SNR [=======                                 ] 3.1069 dB
--------------------------------------------------------------------------------
[2019-03-02 19:53:56] Frame Timing => SEARCHING	 Timing Indicator: STRONG
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 19:53:56] Bit Error Rate: 3.18E-09
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 19:53:56] Carrier Frequency Offset: 29.5730 kHz 
--------------------------------------------------------------------------------
[2019-03-02 19:53:57] SNR [=======                                 ] 3.0847 dB
[2019-03-02 19:53:58] SNR [=======                                 ] 2.9839 dB
[2019-03-02 19:53:59] SNR [=======                                 ] 2.9929 dB
[2019-03-02 19:54:00] SNR [=======                                 ] 3.0035 dB
[2019-03-02 19:54:01] SNR [=======                                 ] 3.1225 dB
[2019-03-02 19:54:02] SNR [=======                                 ] 3.0887 dB
[2019-03-02 19:54:03] SNR [=======                                 ] 3.0921 dB
[2019-03-02 19:54:04] SNR [=======                                 ] 3.0905 dB
[2019-03-02 19:54:05] SNR [=======                                 ] 3.0129 dB
[2019-03-02 19:54:06] SNR [=======                                 ] 2.9666 dB
--------------------------------------------------------------------------------
[2019-03-02 19:54:06] Frame Timing => SEARCHING	 Timing Indicator: STRONG
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 19:54:06] Bit Error Rate: 3.18E-09
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2019-03-02 19:54:06] Carrier Frequency Offset: 29.7354 kHz 
--------------------------------------------------------------------------------
[2019-03-02 19:54:07] SNR [=======                                 ] 3.0451 dB
[2019-03-02 19:54:08] SNR [=======                                 ] 3.1054 dB
[2019-03-02 19:54:09] SNR [=======                                 ] 3.0872 dB
[2019-03-02 19:54:10] SNR [=======                                 ] 3.0173 dB
[2019-03-02 19:54:11] SNR [=======                                 ] 2.9589 dB
[2019-03-02 19:54:12] SNR [=======                                 ] 2.9851 dB
```

blockstreamsatellite, 2019-03-12:
Hello @darosior ,

Would you be so kind to let us know how is your reception now after the [update we released yesterday](https://github.com/Blockstream/satellite#update-on-mar-11-2019). Can you check whether the issue persists? 

Also, if possible, please send screen shots of the "Frame Sync" page in the GUI.

darosior, 2019-03-12:
Thank you for your response. I will try again and let you know.

matthewlilley, 2019-03-15:
Mine looks the same as your second image, everything appears to be working, but I can't find the signal band. I'm using the latest update.

blockstreamsatellite, 2019-03-15:
@matthewlilley Could you send screen shots of the "Frame Sync" page in the GUI? 

Even when the antenna is not pointed correctly, the SNR measurements printed on the console are still non-zero and normally in the range from 1 to 2 dB. All that the receiver is measuring in this case is noise. To confirm that the signal is there, you should either check the "Overview' tab and see the signal band there as in [this plot](https://github.com/Blockstream/satellite/wiki/Understanding-the-GUI#-1-overview) or, better, check that the *Preamble Matched Filter Output* in the Frame Sync page of the GUI shows a peak [like in this example](https://github.com/Blockstream/satellite/wiki/Understanding-the-GUI#-2-frame-synchronization). The peak only appears for the Blockstream Satellite transmission, so it is the way to double check that you are indeed pointed correctly.

matthewlilley, 2019-03-16:
@blockstreamsatellite Yeah, I've included a link below.

Console (on launch) - http://prntscr.com/mypuct

Overview - http://prntscr.com/mypv4k
Frame Sync - http://prntscr.com/mypuub

I don't think it's a positioning problem, I've used the apps, done it manually with a compass etc... Never see anything different, even when manually moving the dish. The only thing I can think of is that maybe the receiver isn't right for this case.

https://www.amazon.co.uk/gp/product/B079C3FHPG/ref=ppx_yo_dt_b_asin_title_o02_s03?ie=UTF8&psc=1

I've never seen the clear band as shown in the Wiki, it seems to just be noise in any position. It doesn't react in an obvious way when I move the dish and am monitoring it in real time. I don't know if that's usual or not.

I'll try to get on the IRC Today.

blockstreamsatellite, 2019-03-16:
@matthewlilley your RTL-SDR is definitely correct. It would be useful to know:

- Your location or region (and the corresponding satellite)
- Your LNB model
- The size of your dish 
- The power injector that you are using

matthewlilley, 2019-03-16:
Thanks again for the quick reply, I really appreciate it.

Location: UK
LNB 1: http://www.inverto.tv/lnb/362/single-universal-40mm-pll-lnb
LNB 2: https://www.amazon.co.uk/gp/product/B07PFC4MFT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
Dish size: 1 meter
Power inserter: https://www.ebay.co.uk/itm/Teleste-IPI-GI-Isolated-Power-Inserter-F-Type-DC-12V-500mA-Free-Delivery-/273656368866

matthewlilley, 2019-03-16:
Both LNBs have a default of low Ku band. One is PPL, and the other is not.

My computed frequency was calculated on the Low LO frequency.

1754020000 = 11504020000 - 9750000000

matthewlilley, 2019-03-17:
The reason I thought it might be the receiver is because it says "capable of frequencies from approximately 25MHz-1700MHz", and it seems the freq I'm using is 1750MHz?

matthewlilley, 2019-03-17:
@blockstreamsatellite Do you see any issues with any of the hardware I linked? I've been out again Today to double check the alignment, and even raised it up another 4ft to be safe, but still nothing.

VDelport, 2019-03-17:
You're right. The RTL-SDR reaches its limits at an input frequency of 1754 MHz. That's why we use the "MK1-PLL Single Std. KU Phase Locked Loop LNBF". The advantage of this LNB is that its LO frequency is 10750 MHz. In this way, the intermediate frequency at the input of the RTL-SDR is exactly 1 GHz lower, ie at about 754 MHz. I hope I could help you with that.

matthewlilley, 2019-03-17:
Thanks @VDelport.

It seems that your LNB is designed for the US region, I also have the option of high band, which would give me a LO of 10750, however, that's unsuitable for my region from what I understand. It seems that the LO for Low Ku band LNBs is always 9750, at least the ones I've looked at. 

I think my only option here might be an extended range RTL SDR?

VDelport, 2019-03-17:
@matthewlilley 
You're right, in Europe we normally use LNBs for DVB reception with a LO frequency of 9.75 GHz for the lower band and a LO frequency of 10.6 GHz for the upper band. That's why we use the mentioned LNB successfully for the reception of Blockstream Satellite in Germany (Mittweida). This LNB does not differentiate between upper and lower band, but always has the same LO frequency of 10.75 GHz.

matthewlilley, 2019-03-17:
Thanks again @VDelport,

This's making a lot more sense. For some reason, I thought I had to use the low band being in the UK.  The LNB you linked is a high band "11.7 - 12.2GHz", which I assumed was incompatible with this Satellite frequency "11.5Ghz". But as it's working for you, it seems not to be the case.

matthewlilley, 2019-03-17:
I'm still a little confused why it works for you though 😆

matthewlilley, 2019-03-17:
[image]

matthewlilley, 2019-03-17:
For now, I've bought an extended range RTL-SDR, since that'll allow me from 65MHz-2300MHz. Hopefully that solves the problem.

blockstreamsatellite, 2019-04-17:
@darosior @matthewlilley do you have updates related to this issue?

@matthewlilley the extended range RTL-SDR solved the problem, right?

FYI, we have now a note in the documentation regarding extended range RTL for Telstar 11N Europe area. See the [SDR requirements](https://github.com/Blockstream/satellite/wiki/Hardware-Requirements#--5-software-defined-radio-interface). Also, there is now a note on the [antenna alignment guide](https://github.com/Blockstream/satellite/wiki/Antenna-Alignment#6-search-for-the-blockstream-satellite-signal) regarding the fact that there is a nearby satellite in Europe that can cause some confusion, as it transmits similar signal bands on similar frequencies.

darosior, 2019-04-18:
> @darosior @matthewlilley do you have updates related to this issue?
>
> @matthewlilley the extended range RTL-SDR solved the problem, right?
>
> FYI, we have now a note in the documentation regarding [...]

We have not looked into this for a while. We'll do as soon as possible and let you know, thank you for your help.

matthewlilley, 2019-04-19:
@blockstreamsatellite  Yes, it was only possible to see the signal band with the extended range RTL-SDR. I did notice the similar band you mention as well. I haven't permanently mounted the dish yet, waiting to get a machine in, once that's done I'll look get it setup again and get back to you.

blockstreamsatellite, 2019-04-19:
Great, @darosior and @matthewlilley , thanks for the feedback.

Do let us know how it goes. We will leave this issue opened in the meantime.

tasoshi, 2019-12-02:
Hello,

_lil update on behalf of @darosior :_
We just tried again to get the signal with blocksat (freshly build from sources)  and failed.
As I just read, the range of my SDR might be not wide enough, and indeed the frequency I need to compute (1754MHz) is just a lil bit above the upperlimit of the SDR (1700MHz)

We are probably going to get a new one to test this out, but i'm still wondering why I get the signal with GQRX at this very frequency.
[image]

blockstreamsatellite, 2019-12-02:
@tasoshi the tuner frequency will set the center frequency, whereas the sampling frequency determines the range of frequencies that can be "observed" around the center frequency. So for example if you are tuned to 1700 MHz and using a sampling frequency of 1 Msps (mega samples per second), you are effectively observing from 1699.5 MHz to 1700.5 MHz (+- 500 kHz around the center frequency). This means that you can observe a signal even if it is beyond the tuner frequency, as long as it falls within the range of frequencies (around the center frequency) determined by the sampling frequency.

By default, `blocksat-rx` uses 1 Msps. However, you can configure it to use a higher sampling frequency. The only drawback is that this will lead to more CPU usage, but if you have enough CPU, I recommend setting this higher sampling frequency when pointing the antenna. To do so, run `blocksat-rx` with argument `-o 8`. This will increase the sampling frequency to 2 Msps. In addition to "observing" a wider range of frequencies, this has the benefit of increase the range of the frequency offset that can be corrected, which can be helpful during pointing.

That all said, I believe your problem is something else. In Europe, as mentioned early, there is a nearby satellite that transmits signals with similar frequency band than T11N Europe (the correct satellite for blocksat). It does seem that you are pointed to the wrong satellite.

Here is a spectrum analyser screenshot and comment posted on satellite's Slack channel by a user a while ago:

[image]

> For anyone else in Europe, above is an overlay of both satellites in close proximity. If you are able to find the 'double signal spectrum' (blue trace) you are not far away. You just need to adjust slightly more to the east, and slightly more elevation.

The signal in black is the correct one. The signal with two similar bands at similar power levels is the wrong one, which seems to be the one on your screenshot.

VDelport, 2019-12-02:
As I mentioned earlier, I recommend using the MK1-PLL Single Std. KU Phase Locked Loop LNBF as the LNB. The advantage of this LNB is that it has one LO frequency, 10750 MHz. In this way, the intermediate frequency at the input of the RTL-SDR is exactly 1 GHz lower, ie at about 754 MHz. In this way, the RTL-SDR is not overwhelmed.
To align the antenna with the Telstar 11 N 37.5 ° W, I recommend first aligning the antenna with the TV satellite Hispasat 30 ° W and checking the correct reception with a DVB-S receiver. After that, the antenna has to be turned a little further in azimuth. Finally, for the best reception, the correct elevation must be jusitized.

pikefloyd, 2019-12-04:
Here is an updated spectrum plot comparing Telstar 11N (BlockSat) along with the spectrum of the satellite east of Telstar 11N.

[image]

If you find yourself on the wrong 'double bump' satellite, (blue trace) as shown above, simply elevate the antenna to reduce the signal levels by 5 or 6dB. Then slowly adjust azimuth to the west by a few degrees and you should see the BlockSat signal, (black trace) start to 'pop up' in the center of the frequency spectrum. Then fine adjust both azimuth, elevation and LNB skew to optimise your signal.

blockstreamsatellite, 2020-05-06:
@darosior FYI we have released a major update on Monday (May 4th 2020), which transitions from our previous protocol to standard DVB-S2. Let us know if you have updates on this ticket. Perhaps you would like to try receiving the new signal?

You can find more information on this [blog post](https://blockstream.com/2020/05/04/en-announcing-blockstream-satellite-2/).

Do note that we will disable the 1.0 network on June 1st 2020.

darosior, 2020-05-07:
Hi,

Thanks. @tasoshi is mostly in charge for this.

> Let us know if you have updates on this ticket

It turned out the issue was with our LNB (our best guess so far), but it's been some time since we tried.

> Perhaps you would like to try receiving the new signal?

I think we'll give the v2 a shot (with a new LNB!) :)

Closing this.

closed 2020-05-07 by darosior