Blockstream/satellite #33: Is the blockstream satellite signal currently being staked out?

VDelport, 2019-03-12:
Hello Blockstream employees,
is the blockstream satellite signal currently being staked out? Although I can see the frequency spectrum in blocksat_gui.rx very well, there is no synchrinization. A few weeks ago, I was able to receive the signal in the best way (MER> 12dB, no block error in several hours of running time). When can we expect the signal emission again? I want to test blocksat on a Raspberry Pi, but without the signal, of course, it will not work. It would be great if you indicated on your website, whether the signal is currently broadcast or not. Then the users could inquire directly there. Thanks in advance.

blockstreamsatellite, 2019-03-12:
Hello @VDelport ,

We updated the uplink signal from our transmitters yesterday, see more information [here](https://github.com/Blockstream/satellite#update-on-mar-11-2019). The update is not backwards compatible with current receiver versions, so you will need to update your software to version v1.3.0. In case you have done so already, let us know and we will investigate your issue further.

By the way, are you on IRC #blockstream-satellite IRC channel on freenode? We can also discuss there.

VDelport, 2019-03-12:
Ok, please let us know how long we can use this update from blocksat to give us some time to work on Blockstream Satellite until the next update.

closed 2019-03-12 by VDelport

blockstreamsatellite, 2019-03-12:
We are not planning on introducing any backwards incompatible change in the near future. Yesterday's update was needed in order to improve reliability problems experienced on weaker coverage areas and setups with low SNR in general. The next planned/upcoming changes will be backwards compatible.

Do let us know in case you have any problems. 

Thanks