Blockstream/satellite #36: 1.4 input to frame synchronizer error

bg002h, 2019-04-16:
Just updated to 1.4. Overview tab of GUI shows incorrect data under the input to frame synchronizer section.  Example: input to frame synchronizer: -0.9 (varies from -1.5 to 10); input to decoder: 10.4.

bg002h, 2019-04-16:
Also, frame sync peak is fluctuating wildly from positive to negative, real to imaginary.

blockstreamsatellite, 2019-04-16:
Hi @bg002h ,

Thanks for commenting.

v1.4.0 introduces some changes that affect the metrics you are referring to. Both results you mentioned are expected and acceptable now. The MER at the input to the frame synchronizer can be substantially low. At the moment, it serves for debugging only. In contrast, the MER at the input to the decoder must be sufficiently high, and it appears only after frame lock.

Regarding the preamble matched filter output of the Frame Sync tab, it is indeed expected to have the peak fluctuating very rapidly from positive to negative, real to imaginary. The block that prevented this behaviour was removed. The decoding process is frame-by-frame, so although it looks like to our own eyes that something is changing rapidly, this phase change is corrected on a frame-by-frame basis (approx 20 times per second) and thus it is acceptable to have the phase of the peak changing over time. What is really desirable is to have a strong peak (positive or negative, real or imaginary), and to have it steadily in the same place (index).

The main `README` explanation is up-to-date with regard to this frame sync behaviour. We will update the Wiki to reflect these changes too.

Other than these metrics, how are the other ones for you? Are you getting frame lock and a reasonable frame error rate?

Thanks again

blockstreamsatellite, 2019-04-17:
@bg002h we have updated the documentation to reflect latest changes and the behaviour you mentioned. You can check for example the [Understanding-the-GUI guide](https://github.com/Blockstream/satellite/wiki/Understanding-the-GUI).

We will close this issue, but do feel free to let us know if you need any help or have any question.

Thanks

closed 2019-04-17 by blockstreamsatellite

bg002h, 2019-04-18:
Thanks for info!