Blockstream/satellite #38: Error building

gitonio, 2019-05-27:
make framers

find: ‘gr-blocksat/lib’: No such file or directory
find: ‘gr-blocksat/lib’: No such file or directory
find: ‘gr-blocksat/lib’: No such file or directory
find: ‘gr-blocksat/include/blocksat’: No such file or directory
find: ‘gr-blocksat/grc’: No such file or directory
find: ‘gr-blocksat/python’: No such file or directory
git submodule update --init gr-framers
mkdir -p gr-framers/build
cd gr-framers/build && cmake .. && make
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   system
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so;/usr/lib/arm-linux-gnueabihf/libgnuradio-pmt.so;/usr/lib/arm-linux-gnueabihf/liblog4cpp.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/arm-linux-gnueabihf/libgnuradio-blocks.so;/usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so;/usr/lib/arm-linux-gnueabihf/libgnuradio-pmt.so;/usr/lib/arm-linux-gnueabihf/liblog4cpp.so
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so;/usr/lib/arm-linux-gnueabihf/libgnuradio-pmt.so;/usr/lib/arm-linux-gnueabihf/liblog4cpp.so
GNURADIO_PMT_FOUND = TRUE
-- Extracting version information from git describe...
-- GNURADIO_PMT_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so;/usr/lib/arm-linux-gnueabihf/libgnuradio-pmt.so;/usr/lib/arm-linux-gnueabihf/liblog4cpp.so
-- Using install prefix: /usr/local
-- Building for version: 7c969ffb / 1.0.0git
--
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- grcc -d /mnt/hdd/blockstream/satellite/gr-framers/build/python /mnt/hdd/blockstream/satellite/gr-framers/python/test_framers.grc
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/hdd/blockstream/satellite/gr-framers/build
make[1]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[2]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[3]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[3]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
[ 18%] Built target gnuradio-framers
make[3]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[3]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
[ 28%] Built target test-framers
make[3]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[3]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
[ 34%] Built target _framers_swig_doc_tag
make[3]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[3]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
make[3]: Entering directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
[ 37%] Generating doxygen xml for framers_swig_doc docs
/bin/sh: 1: ./_framers_swig_doc_tag: Permission denied
swig/CMakeFiles/framers_swig_swig_doc.dir/build.make:64: recipe for target 'swig/framers_swig_doc_swig_docs/xml/index.xml' failed
make[3]: *** [swig/framers_swig_doc_swig_docs/xml/index.xml] Error 126
make[3]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
CMakeFiles/Makefile2:238: recipe for target 'swig/CMakeFiles/framers_swig_swig_doc.dir/all' failed
make[2]: *** [swig/CMakeFiles/framers_swig_swig_doc.dir/all] Error 2
make[2]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
Makefile:140: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/mnt/hdd/blockstream/satellite/gr-framers/build'
Makefile:98: recipe for target 'build/gr-framers_build_record' failed
make: *** [build/gr-framers_build_record] Error 2

blockstreamsatellite, 2019-05-27:
If possible, please send us more information about the build environment.

- What is the distribution?
- Are you using the head of master (latest commit) on all repositories (`satellite`, `gr-blocksat` and `gr-framers`)?
- Is this an ARM? If yes, is it 32 bit or 64 bit? 

It seems you have a problem with directory permissions, based on `/bin/sh: 1: ./_framers_swig_doc_tag: Permission denied`. Maybe related to not having enough permissions on `/mnt/hdd/`?

Do note that by default the installation prefix is `/usr/local`. If you don't want to install it there (maybe because you want somewhere around `/mnt/hdd`), you can change this by overriding `DESTDIR` on `make install`, like so:

```
make install DESTDIR=your_target_install_directory
```

gitonio, 2019-05-27:
Thanks for your response.

root@odroid:/mnt/hdd/blockstream/satellite/gr-framers# uname -a
Linux odroid 4.14.111-158 #1 SMP PREEMPT Tue Apr 16 12:26:31 -03 2019 armv7l armv7l armv7l GNU/Linux

64 bit

root@odroid:/mnt/hdd/blockstream/satellite/gr-framers# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

Did: root@odroid:/mnt/hdd/blockstream# git clone https://github.com/Blockstream/satellite.git

Regarding permissions: Running as root. Root owns /mnt/hdd.

blockstreamsatellite, 2019-05-27:
Can you share the output of `lscpu`? ARMv7 is 32-bit, isn't it? The reason for the question is that we haven't had comprehensive tests of building under 32-bit yet. Still, it could just work.

Another question is about your GNU Radio version. Can you run the following and share the results?

```
gnuradio-config-info --version
```

You may try to build `gr-framers`, `gr-blocksat` and `satellite` independently. Also, you may want to disable `doxygen` documentation to avoid the problem you are having, which apparently happens when the doxygen documentation is generated. 

### gr-framers:

```
git clone https://github.com/gr-vt/gr-framers.git
cd gr-framers
mkdir build
cd build
cmake -DENABLE_DOXYGEN=OFF ..
make
make install
```
Or `sudo make install`. To set the installation destination, you can add the following flag to the `cmake` step: `-DCMAKE_INSTALL_PREFIX=target_dir`.

### gr-blocksat

The same processs:
```
git clone https://github.com/Blockstream/gr-blocksat.git
cd gr-blocksat
mkdir build
cd build
cmake -DENABLE_DOXYGEN=OFF ..
make
make install
```

### satellite

```
git clone https://github.com/Blockstream/satellite.git
cd satellite
make
make install
```

gitonio, 2019-05-27:
root@odroid:/mnt/hdd/blockstream/satellite/gr-framers# lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              8
On-line CPU(s) list: 0-7
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           2
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A7
Stepping:            r0p3
CPU max MHz:         2000.0000
CPU min MHz:         200.0000
BogoMIPS:            18.00
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae

root@odroid:/mnt/hdd/blockstream/satellite/gr-framers# gnuradio-config-info --version
3.7.11

blockstreamsatellite, 2019-06-02:
@gitonio did the above build steps (including `cmake -DENABLE_DOXYGEN=OFF`) work for you?

Let us know if you need further help. Thanks

gitonio, 2019-06-30:
The doxygen problem went away. However still have the permissions issue:

root@odroid:/mnt/hdd/blockstream/satellite/gr-framers/build# make
[ 22%] Built target gnuradio-framers
[ 33%] Built target test-framers
[ 33%] Built target framers_swig_swig_doc
[ 40%] Built target _framers_swig_swig_tag
[ 44%] Generating framers_swig.tag
/bin/sh: 1: ./_framers_swig_swig_tag: Permission denied
swig/CMakeFiles/framers_swig_swig_2d0df.dir/build.make:96: recipe for target 'swig/framers_swig.tag' failed
make[2]: *** [swig/framers_swig.tag] Error 126
CMakeFiles/Makefile2:305: recipe for target 'swig/CMakeFiles/framers_swig_swig_2d0df.dir/all' failed
make[1]: *** [swig/CMakeFiles/framers_swig_swig_2d0df.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

gitonio, 2019-09-23:
After moving the git clone to a different mount point it works according to the readme directions. Thanks for the help.

closed 2019-09-23 by gitonio