Blockstream/satellite #39: Decode data from Satellite API

whiteyhat, 2019-05-27:
How can I decode the message_digest from the blockstream satellite API? 

I have read the python example but it is not very clear. Is it any JavaScript available to make the transmission readable?

blockstreamsatellite, 2019-05-29:
@whiteyhat, thanks for your question.

 In the most typical receiver use case, there is no need to interact with the API. The receiver application (`blocksat-rx`) receives API messages and the `api_data_reader.py` example script tries to [decrypt the data](https://github.com/Blockstream/satellite/blob/master/api/examples/api_data_reader.py#L261). The idea is that decrypting will only be successful for the messages that you are really the target recipient of (you have the GPG keys to decrypt).

Is that what you mean by "decode" (to decrypt)? If not, could you clarify the question?

Let us know.

bokobza, 2019-05-29:
@whiteyhat the message_digest in `GET /order/:uuid`  is just a SHA256 hash of the message or file that was sent when creating an order, so it's not there to be decrypted (you can't, anyway).

If you want to see a broadcasted message, you have to use the `/message` endpoint, for example `https://api.blockstream.space/message/{seq_num}`, seq_num being a number you can find when you query `GET /order/:uuid`.

whiteyhat, 2019-05-30:
Interesting! Okay, that is all I needed thank you guys! @bokobza @blockstreamsatellite 

One last thing, querying the Satellite API requires internet access?  Eg:

`curl -F "bid=10000" -F "message=Hello World" https://api.blockstream.space/order/`

bokobza, 2019-05-30:
No problem.
Yeah you need internet to query the API. The API is the one that push messages to the satellite.

closed 2019-05-31 by whiteyhat