Blockstream/satellite #42: How do you run blocksat-rx in the background?

VDelport, 2020-02-28:
For a long-term test, we would like to run blocksat-rx in the background. After the command "blocksat-rx - f ... -g ... &" blocksat-rx starts successfully, but is then ended, apparently because a terminal input is promptly queried.

blockstreamsatellite, 2020-02-28:
Hi @VDelport , 

You are right. This is a known behavior, but which is configurable. A GNU Radio can be generated both in a mode on which the user is prompted for exit, as well as in a mode where the app runs until completion (or explicit user termination).

We've just pushed a commit (19afecb) that changes the mode. After pulling the changes, run `make` and `make install` again.

Let us know if it helps.

VDelport, 2020-03-01:
Thank you for the quick reply and the hint. It works.

closed 2020-03-02 by blockstreamsatellite