Blockstream/satellite #46: Bitcoind: Network is unreachable

reynico, 2020-03-16:
Hi!
I'm running Blockstream Satellite and Bitcoinfibre on a Raspberry PI 3 with Ubuntu Mate.

I think the decoding part is doing well, here's some logging from `blocksat-rx`

```
nico@sdr:~$ sudo blocksat-rx -f 1426400000 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[2020-03-16 20:20:54] SNR [=                                       ] 0.0000 dB
Press Enter to quit: [2020-03-16 20:20:55] SNR [==========                              ] 4.3307 dB

##########################################
-- Frame synchronization acquired
-- On Mon Mar 16 20:20:55 2020
##########################################

[2020-03-16 20:20:56] SNR [==============                          ] 6.5202 dB
[2020-03-16 20:20:57] SNR [==============                          ] 6.2754 dB
[2020-03-16 20:20:58] SNR [==============                          ] 6.5336 dB
[2020-03-16 20:20:59] SNR [==============                          ] 6.5540 dB
[2020-03-16 20:21:00] SNR [==============                          ] 6.3109 dB
--------------------------------------------------------------------------------
[2020-03-16 20:21:00] Frame Timing => LOCKED	 Timing Indicator: GOOD
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2020-03-16 20:21:00] Frame Error Rate: 1.52E-02   1 dropped /    66 frames
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2020-03-16 20:21:00] Carrier Frequency Offset: 76.2939 kHz 
--------------------------------------------------------------------------------
[2020-03-16 20:21:01] SNR [==============                          ] 6.3117 dB
[2020-03-16 20:21:02] SNR [=============                           ] 6.2427 dB
[2020-03-16 20:21:03] SNR [==============                          ] 6.3340 dB
[2020-03-16 20:21:04] SNR [==============                          ] 6.3688 dB
[2020-03-16 20:21:05] SNR [=============                           ] 6.1948 dB
[2020-03-16 20:21:06] SNR [=============                           ] 6.2475 dB
[2020-03-16 20:21:07] SNR [=============                           ] 6.2311 dB
[2020-03-16 20:21:08] SNR [==============                          ] 6.2580 dB
[2020-03-16 20:21:09] SNR [==============                          ] 6.3543 dB
[2020-03-16 20:21:10] SNR [==============                          ] 6.2992 dB
--------------------------------------------------------------------------------
[2020-03-16 20:21:10] Frame Timing => LOCKED	 Timing Indicator: GOOD
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2020-03-16 20:21:10] Frame Error Rate: 4.95E-03   1 dropped /   202 frames
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2020-03-16 20:21:10] Carrier Frequency Offset: 76.2939 kHz 
--------------------------------------------------------------------------------
[2020-03-16 20:21:11] SNR [==============                          ] 6.4068 dB
[2020-03-16 20:21:12] SNR [==============                          ] 6.5256 dB
[2020-03-16 20:21:13] SNR [==============                          ] 6.4050 dB
[2020-03-16 20:21:14] SNR [=============                           ] 6.1817 dB
--------------------------------------------------------------------------------
[2020-03-16 20:21:14] Rx Data	Blocks: 247.0 kB (13.6 kB/sec)
                             	   API:      0 B (0.0 kB/sec)
[... 226 lines cut ...]
2020-03-16 23:27:05 connect() to [2001:0:9d38:6ab8:30e7:3e5c:da3f:6466]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:05 connect() to [2001:0:9d38:6ab8:1498:36b6:a29b:12a4]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:06 connect() to 81.228.194.187:8333 failed: Network is unreachable (101)
2020-03-16 23:27:06 connect() to [2001:0:5ef5:79fb:2853:3b95:a430:40f3]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:07 connect() to [2a02:c205:2009:7804::1]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:07 connect() to [2607:5300:60:9ad::1]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:08 connect() to [2002:2df9:5e40::2df9:5e40]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:08 connect() to [2a01:4f8:140:5329::102]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:09 connect() to [2001:0:9d38:6ab8:2015:473:9d21:cc3c]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:09 ERROR: AcceptBlockHeader: prev block not found
2020-03-16 23:27:09 ERROR: ProcessNewBlock: AcceptBlock FAILED ()
2020-03-16 23:27:09 UDP: Failed to decode block 000000000000000000039eccbd9fb5da94c3a8efe34fa6fc8cc207e2b69fcda6
2020-03-16 23:27:09 connect() to [2001:0:9d38:953c:3c43:3f1:b489:f8b8]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:10 connect() to [2a00:c440:10:1c62:1234::1]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:10 connect() to [2001:0:4137:9e76:20ac:22b1:bb18:1160]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:11 connect() to 104.156.99.240:8333 failed: Network is unreachable (101)
2020-03-16 23:27:11 connect() to [2605:9880:201:17::4b7c]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:12 connect() to 188.234.241.172:8333 failed: Network is unreachable (101)
2020-03-16 23:27:12 connect() to [2001:0:9d38:6ab8:3439:223:a04a:fdfd]:8333 failed: Network is unreachable (101)
2020-03-16 23:27:13 connect() to 94.237.64.138:8333 failed: Network is unreachable (101)
```

and the `failed: Network is unreachable (101)` errors just repeats over and over.

What am I missing here?

VDelport, 2020-03-17:
Start bitcoind once with the additional option -connect = 0. With this option, bitcoind does not attempt to connect to a node on the Internet.

reynico, 2020-03-17:
Thanks! I'll give a try today.

closed 2020-03-17 by reynico