Blockstream/satellite #47: ldpc_tool compilation error

634-5789, 2020-05-08:
`make CXX=g++ ldpc_tool`

fails with

/usr/bin/ld: mods_handler.o:mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5softNEPaPS2_f[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5softNEPaPS2_f]+0x64): more undefined references to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:18: ldpc_tool] Error 1

on Ubuntu 19.10. How can this be fixed?

blockstreamsatellite, 2020-05-08:
Thanks for letting us know. We have been able to reproduce the problem.

Could you try again by repeating the process? We have pushed some changes to the `leansdr` repository as well as the underlying `LDPC` submodule. Hence, you could remove the `leansdr` folder, then clone and build again. Alternatively, you could run:

```
cd leansdr/
git pull origin master
git submodule update
```

634-5789, 2020-05-08:
That worked. Thanks for the quick response and keep up this great work.

closed 2020-05-08 by 634-5789

blockstreamsatellite, 2020-05-08:
@634-5789 You are welcome. Thanks for confirming it works.