Blockstream/satellite #49: Installation of TSDuck from the sources

VDelport, 2020-06-04:
Hello, I'm trying to install the new version of blocksat, for reception with the RTL-SDR and by compiling the sources. The TSDuck installation guide states:

'And then add the following to your .bashrc:
source ~ / .blocksat / src / tsduck / src / tstools / release-x86_64 / setenv.sh'

But this directory does not exist or is not created.

On the TSDuck website (https://tsduck.io/doxy/building.html) it says: 'The TSDuck binaries, executables and shared objects (.so), are built in directories bin/release-i386-hostname, bin/release-x86_64-hostname and bin/release-arm-hostname for Intel 32-bit, 64-bit and ARM platforms respectively. ' 

It is the same after my installation.

blockstreamsatellite, 2020-06-05:
@VDelport Thanks for reporting the issue.

There was a recent update on TSDuck that changed the paths where the `setenv.sh` script is located. To cope with that, we have recently pushed an update changing the paths where we expect to place binaries. The update was on commit https://github.com/Blockstream/satellite/commit/e791ba3302abc49fcdc557b07815b9da3ba12684.

Based on the fact that you are experiencing this issue, I assume you didn't use the automatic installation of dependencies using command `blocksat-cli deps install` and instead tried to compile and install from source manually. Unfortunately, the manual compilation instructions were missing the path updates. We have just updated them, and you can see them at https://github.com/Blockstream/satellite/blob/master/doc/sdr.md#further-information

The main modification is that binaries are now expected to be at `~/.blocksat/usr/bin` instead of `~/.blocksat/bin`. Furthermore, we don' rely on the `setenv.sh` script from TSDuck anymore. So if you did add that to your `~/.bashrc`, you can remove it now. Instead of relying on this script, now we actually install TSDuck into `~/.blocksat/usr`, by running an extra step that didn't exist in the instructions before:
```
make install SYSPREFIX=$HOME/.blocksat/usr
```

Note the path changes are also the reason why you have the problem reported in #50 .

If you want a quick solution, I would recommend the following sequence:

1. Remove the `~/.blocksat/bin` directory that is not used anymore: `rm -r ~/.blocksat/bin`
2. Remove the `setenv.sh` line from your `.bashrc`, as it is also not used anymore
3. Run `blocksat-cli deps install`

Alternatively, instead of running step 3, manually run only the installation steps again, as follows:

```
mkdir -p ~/.blocksat/usr/bin
cd ~/.blocksat/src/leansdr/src/apps
install leandvb ~/.blocksat/usr/bin
cd ../../LDPC/
install ldpc_tool ~/.blocksat/usr/bin
cd ~/.blocksat/src/tsduck
make install SYSPREFIX=$HOME/.blocksat/usr
```

Note that command `blocksat-cli deps install` runs the sequence in https://github.com/Blockstream/satellite/blob/master/blocksatcli/dependencies.py#L25, so it is essentially the same as building and installing everything manually.

VDelport, 2020-06-05:
@blockstreamsatellite 
Thank you. 
After make install SYSPREFIX=$HOME/.blocksat/usr follows this:
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
make -C src install
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
  [DEP] dtv/static/tsStaticReferencesDVB.cpp
  [DEP] tsduck.cpp
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
install -d -m 755 /home/delport/.blocksat/usr/bin
rm -f \
    /home/delport/.blocksat/usr/bin/tsduck.xml \
    /home/delport/.blocksat/usr/bin/tsduck.channels.xml \
    /home/delport/.blocksat/usr/bin/tsduck.dvb.names
install -m 644 \
    /home/delport/.blocksat/src/tsduck/bin/release-x86_64-pc_name/tsduck.so \
    dtv/tsduck.keytable.model.xml dtv/tsduck.hfbands.xml dtv/tsduck.lnbs.model.xml dtv/tsduck.channels.model.xml dtv/tsduck.hfbands.model.xml dtv/tsduck.lnbs.xml dtv/tsduck.tables.model.xml dtv/tsduck.names dtv/tsduck.oui.names \
    /home/delport/.blocksat/usr/bin
/bin/bash: curl-config: command not found
/bin/bash: curl-config: command not found
  [CXX] tsplugin_teletext.cpp
In file included from /home/delport/.blocksat/src/tsduck/src/libtsduck/base/tsUChar.h:37:0,
                 from /home/delport/.blocksat/src/tsduck/src/libtsduck/base/tsUString.h:36,
                 from /home/delport/.blocksat/src/tsduck/src/libtsduck/base/tsReport.h:36,
                 from /home/delport/.blocksat/src/tsduck/src/libtsduck/base/tsArgs.h:36,
                 from /home/delport/.blocksat/src/tsduck/src/libtsduck/plugin/tsPlugin.h:36,
                 from /home/delport/.blocksat/src/tsduck/src/libtsduck/plugin/tsInputPlugin.h:36,
                 from /home/delport/.blocksat/src/tsduck/src/libtsduck/plugin/tsPluginRepository.h:36,
                 from tsplugin_teletext.cpp:35:
/home/delport/.blocksat/src/tsduck/src/libtsduck/base/tsPlatform.h:871:10: fatal error: winscard.h: No such file or directory
 #include <winscard.h>
          ^~~~~~~~~~~~
compilation terminated.
../../Makefile.common:936: recipe for target '/home/delport/.blocksat/src/tsduck/bin/release-x86_64-pc_name/objs-tsplugins/tsplugin_teletext.o' failed
make[2]: *** [/home/delport/.blocksat/src/tsduck/bin/release-x86_64-pc_name/objs-tsplugins/tsplugin_teletext.o] Error 1
Makefile:47: recipe for target 'install' failed
make[1]: *** [install] Error 2
Makefile:90: recipe for target 'install' failed
make: *** [install] Error 2

blockstreamsatellite, 2020-06-05:
@VDelport When running the instructions for the first time, did you run the step below?

```
build/install-prerequisites.sh
```

Note this should be executed after `cd ~/.blocksat/src/tsduck`. See the [SDR guide](https://github.com/Blockstream/satellite/blob/master/doc/sdr.md#further-information).

This script (`install-prerequisites.sh`) should install the required libraries. It seems the one that is missing is `libpcsclite-dev` (in Ubuntu).

VDelport, 2020-06-05:
@blockstreamsatellite 
>When running the instructions for the first time, did you run the step below?
>build/install-prerequisites.sh
Yes I did. I use Linux Mint.

blockstreamsatellite, 2020-06-06:
@VDelport Right. See if you can install the equivalent of `libpcsclite-dev` on Linux Mint.

You can see the list of packages that are installed by the `install-prerequisites.sh` script in the following link:

https://github.com/tsduck/tsduck/blob/master/build/install-prerequisites.sh#L123

VDelport, 2020-06-11:
bash -x build/install-prerequisites.sh
++ basename build/install-prerequisites.sh .sh
+ SCRIPT=install-prerequisites
+++ dirname build/install-prerequisites.sh
++ cd build
++ pwd
+ SCRIPTDIR=/home/delport/.blocksat/src/tsduck/build
+ STATIC=false
+ M32=false
+ [[ 0 -gt 0 ]]
++ lsb_release -i
++ sed -e 's/.*:[\t ]*//'
+ DISTRO=LinuxMint
++ lsb_release -r
++ sed -e 's/.*:[\t ]*//' -e 's/\..*//'
+ MAJOR=19
++ lsb_release -r
++ sed -e '/\./!d' -e 's/.*:[\t ]*//' -e 's/.*\.//'
+ MINOR=
++ uname -s
+ [[ Linux == Darwin ]]
+ [[ LinuxMint == \U\b\u\n\t\u ]]
+ [[ LinuxMint = \D\e\b\i\a\n ]]
+ [[ LinuxMint = \R\a\s\p\b\i\a\n ]]
+ [[ -f /etc/fedora-release ]]
+ [[ -f /etc/redhat-release ]]
+ [[ -f /etc/arch-release ]]
+ [[ -f /etc/alpine-release ]]

lsb_release -i
#Distributor ID:	LinuxMint

lsb_release -r
Release:	19

blockstreamsatellite, 2020-06-11:
Thanks for the information, @VDelport So in this case, with the given `lsb_release` outputs, the `install-prerequisites.sh` script really won't install anything (and the logs confirm).

What is your package manager? If `apt`, you can run:

```
apt install g++ dos2unix curl tar zip doxygen graphviz pcscd libpcsclite-dev dpkg-dev
```

After installing the above packages, what happens if you run the following?

```
cd ~/.blocksat/src/tsduck
build/install-prerequisites.sh
make NOTELETEXT=1 NOSRT=1 NOPCSC=1 NOCURL=1 NODTAPI=1
```

If everything goes well, you can proceed to the installation step (below). If not, please share the error logs so that we can troubleshoot further.

```
make install SYSPREFIX=$HOME/.blocksat/usr
```

VDelport, 2020-06-11:
Thank you, the binaries from tsduck were then successfully installed in .blocksat / usr / bin. The dos2unix and libpcsclite-dev packages were not previously installed.

closed 2020-06-12 by VDelport