Blockstream/satellite #51: Please provide a transport stream file with embedded Bitcoin blocks

VDelport, 2020-06-11:
Since I currently have no access to our satellite reception system, I would like to install my own DVB-S2 transmitter with an SDR device in order to test blocksat-cli sdr. Would you please provide a transport stream file * .ts with one or two embedded Bitcoin blocks for such tests, which corresponds to the format that you send out via satellite?

blockstreamsatellite, 2020-06-15:
@VDelport Are you looking for a TS file or an IQ file?  There is currently no way to feed a TS file into `blocksat-cli sdr`, although we can add one in the future.

To clarify, an IQ file contains the IQ samples that the RTL-SDR would normally provide to the software-based demodulator. This is generally a very large file. To record an IQ file, one can use:

```
blocksat-cli sdr rec
```

To read an IQ file and demodulate the signal as if it was coming from an RTL-SDR, the command is:

```
blocksat-cli sdr -f [iq_file]
```
where `[iq_file]` is the path to the IQ file.

In contrast, a TS file contains the MPEG TS packets that are output after demodulation/decoding of the DVB-S2 signal. Currently, it's possible to record a TS file via `blocksat-cli` using a USB demodulator (i.e. currently the TBS5927 demodulator). In particular, with command:

```
blocksat-cli usb launch -r [ts_file]
```
where `[ts_file]` is the path to the desired TS file.

We will notify you once we add to support to recording and loading of TS files on the `sdr` command.

VDelport, 2020-06-15:
Thank you for your detailed answer. But as I said, I cannot currently receive a satellite signal. That's why I built my own DVB-S2 transmission system. To test the blocksat-cli sdr, a .ts file with embedded blockchain data would be useful.

blockstreamsatellite, 2020-06-26:
@VDelport Here is an example TS file (attached). I assume that's what you looking for, rather than an IQ file. However, as stated earlier, note that the current version of the `blocksat-cli` does not support the injection of data from a TS file.

Once you unzip the TS file, you could inject the data into bitcoin-satellite by using the following `tsp` command:
```
tsp -I file blocksat.ts -P mpe --pid 32-33 --udp-forward - -local-address 127.0.0.1 -O drop
```
> Note that `tsp` is part of the tsduck package. It may not be available globally in your system if you simply installed tsduck at `~/.blocksat/usr/bin`, according to the former version of the sdr installation instructions. So you can call `tsp` directly from this directory. However, note that the most up-to-date version of the `blocksat-cli` installs tsduck globally instead, via binary packages (available for Ubuntu, Fedora, and CentOS only) that we prepared recently. If you are running one these distributions, you could upgrade the cli and start using the binary packages. To do so, you would need to run `sudo pip3 install blocksat-cli --upgrade`, followed by `blocksat-cli deps install`.

Furthermore, note that your `bitcoin.conf` needs the configuration that is done via `blocksat-cli btc` command for an SDR setup, which adds option `udpmulticast` listening to the loopback interface (`lo`).  The above `tsp` command will forward the UDP data decapsulated from the MPEG TS stream into the loopback interface (address 127.0.0.1).

[blocksat.ts.zip](https://github.com/Blockstream/satellite/files/4839526/blocksat.ts.zip)

VDelport, 2020-06-28:
Thank you for your support and the provision of the transport stream files with the embedded Bitcoin blocks 636497-636499.

closed 2020-06-28 by VDelport