Blockstream/satellite #53: Bitcoin satellite shuts down due to missing UDP connection

VDelport, 2020-06-16:
Hello, I'm trying to receive the Bitcoin blocks with blocksat-cli sdr. With Wireshark I can record that the UPD segments arrive at localhost (127.0.0.1).

I start bitcoin satellite with the following command line:

bitcoind -printtoconsole -connect = 0

The file .bitcoin.conf has the following content:
debug = udpnet
debug = udpmulticast
udpmulticastloginterval = 10
udpmulticast = lo, 239.0.0.2: 4434,127.0.0.1,1, blocksat-sdr

bitcoind shuts down with the error:
2020-06-16T12: 41: 22Z Error: Failed to initialize UDP connections
Error: Failed to initialize UDP connections

Which other configurations have to be carried out on the computer so that the UDP connection can be established?

blockstreamsatellite, 2020-06-16:
Hi @VDelport ,

Is your loopback interface named `lo`? If not, please change the first parameter on option `udpmulticast` (currently set to `lo`) in your `bitcoin.conf` file. 

If that's not the problem, could you please provide further logs? Especially the logs that are printed before `Failed to initialize UDP connections`.

VDelport, 2020-06-16:
Thanks for the hint, I also had to run:

sysctl -w net.core.rmem_max = 24660008

closed 2020-06-16 by VDelport

blockstreamsatellite, 2020-06-16:
@VDelport Correct. Configuration `net.core.rmem_max` must be set so that the socket buffers have more room for the packets coming via satellite. If the user's `rmem_max` is inferior to our requested minimum (24660008), `bitcoind` will print this recommendation (to set `net.core.rmem_max`) and exit.