Blockstream/satellite #54: Bitcoin Satellite cannot be built on a 32-bit Debian.

VDelport, 2020-06-25:
Hello, I'm trying to build your version of Bitcoin Fibre, which you call Bitcoin Satellite, on a 32-bit computer with a Debian operating system (4.9.0-9-686-pae # 1 SMP Debian 4.9.168-1 + deb9u3 (2019-06-16) i686 GNU / Linux). 

I get the following error message:

In file included from wirehair / gf256.h: 70: 0,
                  from wirehair / gf256.cpp: 30:
/usr/lib/gcc/i686-linux-gnu/6/include/tmmintrin.h:136:1: error: inlining failed in call to always_inline ’__m128i _mm_shuffle_epi8 (__ m128i, __m128i)’: target specific option mismatch
  _mm_shuffle_epi8 (__m128i __X, __m128i __Y)

What compiler options do I need to use to configure the build process?

By the way, this error does not occur when creating the original version of Bitcoin Fibre.

Can the original version of Bitcoin Fibre also synchronize the Bitcoin blocks that were transmitted with the DVB-S2 receiving system?

blockstreamsatellite, 2020-06-25:
@VDelport could you post the issue on https://github.com/Blockstream/bitcoinsatellite

Also, when you do, please let us know more about the specific distribution and release so that we can reproduce the problem.

VDelport, 2020-06-26:
Unfortunately I can't find any tab Issues on bitcoinsatellite. A Debian operating system (4.9.0-9-686-pae # 1 SMP Debian 4.9.168-1 + deb9u3 (2019-06-16) i686 GNU / Linux is running on the computer.

blockstreamsatellite, 2020-06-26:
@VDelport Thanks for letting us know. Issues tab is active there now.

closed 2020-06-26 by blockstreamsatellite