Blockstream/satellite #55: Please increase the transmission power of the DVB-S2 transmitter

VDelport, 2020-06-28:
I have now carried out the first tests with our satellite reception system in Mittweida (Germany, Saxony). I receive the DVB-S2 signal for Europe with a 90 cm satellite dish and the RTL-SDR device.

In principle, reception is possible in both modes (low throughput and high throughput). I was able to receive blocks in both modes. However, the received signal is quite weak (see screenshot from GQRX). The MER is around 9 dB, which I consider borderline (see screenshot leandvb). I am therefore afraid that in poor reception conditions (weather) the necessary carrier-to-noise ratio (CNR) is no longer sufficient to demodulate the DVB-S2 signal.

Is it possible that you increase the DVB-S2 transmission signal by approx. 10 dB?

[image]

[image]

blockstreamsatellite, 2020-06-29:
Hi @VDelport 

Thanks for reporting your results. 

> However, the received signal is quite weak (see screenshot from GQRX). The MER is around 9 dB, which I consider borderline (see screenshot leandvb).

Your MER (around 9 dB) is within the level that we recommend and expect for a 90cm dish. You can find more information here: https://github.com/Blockstream/satellite/blob/master/doc/antenna-pointing.md#optimize-snr

>  I am therefore afraid that in poor reception conditions (weather) the necessary carrier-to-noise ratio (CNR) is no longer sufficient to demodulate the DVB-S2 signal.

Note that blocksat 2.0 transmits DVB-S2 in the so-called VCM (variable coding and modulation) mode, in particular by multiplexing two modulation and coding (MODCOD) schemes over time. One MODCOD is quite reliable (has minimum SNR of 1 dB only), and that's the one that carries our low-throughput (~96 kbps) stream. The other MODCOD is meant for higher-throughput, and so it is not as robust as the other one, but carries more data per second (1.55 Mbps). Since the two streams are continuously multiplexed in the same carrier, the idea is that, under poor weather conditions, a VCM capable receiver still likely maintains communication through the low-throughput stream, even if it stops receiving the high-throughput stream. Moreover, note that new blocks are broadcast in bitcoin satellite through both the low and high throughput streams. Hence, under poor weather conditions, the user is still likely to remain in sync by receiving the new blocks through the low-throughput stream.

Nevertheless, the SDR setup has the limitation in that it can only receive the slow or the fast stream, but not both simultaneously. It's in our roadmap to overcome this limitation, as it is a software limitation. However, in the current version, the MODCOD still needs to be selected between the two. In contrast, the USB (TBS5927) and standalone (Novra S400) modems do not have this limitation. These two devices can receive the slow and fast streams simultaneously. 

Thus, on an SDR setup, if your goal is to maintain connectivity as much as possible, I'd recommend running with the slow stream (the default). In contrast, if maximum throughput is the goal despite the potential outages in poor weather, then you can try the high-throughput MODCOD (option `-m high`). You could even try to toggle the two modes on your end depending on whether the receiver can lock to the high-throughput stream. 

> Is it possible that you increase the DVB-S2 transmission signal by approx. 10 dB?

I'm afraid it's not possible. Satellite communications services in general operate at a specified power level, which has to be maintained continuously. However, as mentioned eariler, your signal power is within the expected range. Hence, you can expect reasonable performance. If you are not receiving data well enough, please refer to the further notes given below.

If you want to assess your current performance in terms of error rate at MPEG TS level, I'd recommend running `blocksat-cli sdr` command with argument `--monitor-ts`. That will show whenever there is a gap on the MPEG TS packet sequence number. Alternatively, you can run in debug mode, by appending command-line argument `-d`. In this mode, the leandvb application will print an underscore for each succesfully recovered DVB-S2 frame packet and an exclamation mark `!` for each DVB-S2 frame with remaining errors.

A very important note is that reception of the high throughput stream is challenging in terms of CPU utilization. If you see frequent errors when monitoring through the above options (e.g., with `--monitor-ts`), the first recommendation would be to check your CPU utilization. If it's close to 100%, it means that the errors you are getting are mostly from overflows (data loss) due to CPU not keeping up with the pace required for processing. In this case, you can play with the command-line option `--ldpc-iterations`. The default number of iterations is 25. You can try to reduce this number so that the CPU utilization reduces.

This option `--ldpc-iterations` determines the maximum number of iterations that the LDPC decoder will use when trying to correct the errors in the incoming DVB-S2 frames. After this number of iterations, the decoder gives up and leaves the decoded frame with remaining errors. By default, the application will run six LDPC decoder threads (also configurable via option `-n`), each configured for a maximum of 25 iterations. In general, the higher the number of iterations, the better for decoding performance. However, at some point the number of interations is infeasible for the CPU, such that the application won't be able to keep up with the pace of the incoming data. In this scenario, the application will lose some data from the RTL, which then typically leads to the leandvb application unlocking and losing even more data. So the bottom line is that the number of iterations should never be higher that what your CPU can handle safely, otherwise the performance deteriorates significantly. The number of iterations requires some tuning on each specific host/setup.

blockstreamsatellite, 2020-07-27:
@VDelport let us know if we can close this ticket.

closed 2021-05-18 by blockstreamsatellite