Blockstream/satellite #56: No DVB-S2 adapters found

VDelport, 2020-07-10:
I am trying to run blocksat-cli in the variant TSP 5927 on Ubuntu 20.04 LTS in a VM and fail because the DVB-S2 adapter is not found, although the VM recognizes the USB interface of the TSP 5927 and I also selected the USB interface.

sudo blocksat-cli usb config

------------------------------- Find DVB adapter ---------------- ---------------
Traceback (most recent call last):
  File "/ usr / local / bin / blocksat-cli", line 8, in <module>
    sys.exit (main ())
  File "/usr/local/lib/python3.8/dist-packages/blocksatcli/main.py", line 60, in main
    args.func (args)
  File "/usr/local/lib/python3.8/dist-packages/blocksatcli/usb.py", line 564, in usb_config
    common_params = _common (args)
  File "/usr/local/lib/python3.8/dist-packages/blocksatcli/usb.py", line 549, in _common
    adapter, frontend = _find_adapter ()
  File "/usr/local/lib/python3.8/dist-packages/blocksatcli/usb.py", line 155, in _find_adapter
    assert (len (dvb_s2_adapters)> 0), "No DVB-S2 adapters found"
AssertionError: No DVB-S2 adapters found

blockstreamsatellite, 2020-07-10:
Hi @VDelport ,

Have you installed the drivers for the TBS5927? See here: https://github.com/Blockstream/satellite/blob/master/doc/tbs.md#tbs-5927-drivers

If you have installed the drivers already, can you see any `dvb` interface at `/dev/` ? Check `ls /dev/dvb*`.

blockstreamsatellite, 2020-07-10:
Also, @VDelport , do note that you need to upgrade the CLI to the latest version (0.2.5) in order to run command `blocksat-cli deps tbs-drivers`. You can upgrade by running:

```
sudo pip3 install blocksat-cli --upgrade
```

Here is a quick reference guide for CLI commands in case it's useful to you: https://github.com/Blockstream/satellite/blob/master/doc/quick-reference.md

VDelport, 2020-07-10:
>Have you installed the drivers for the TBS5927?
I have run blocksat-cli deps tbs-drivers and reboot.
>If you have installed the drivers already, can you see any dvb interface at /dev/ ? Check ls /dev/dvb*.
There is no /dev/dvb*

blockstreamsatellite, 2020-07-10:
@VDelport so here are few other things you can try.

Check if the drivers are installed by running:
```
modinfo dvb_usb_tbs5927
```
or
```
find /lib/modules/$(uname -r)/kernel/drivers/ -iname "dvb-usb-tbs5927.ko"
```

If the drivers are installed (i.e., you found it with the above commands), you can see they are being used:
```
lsmod | grep dvb_usb_tbs5927
```

Also, you can check `dmesg` and look for logs like so:
```
usb 1-1.3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5927 DVB-S/S2).
```

VDelport, 2020-07-10:
Thank you, now the DVB-S2 adapter was detected.

blockstreamsatellite, 2020-07-10:
Great, @VDelport . Let us know if you have further questions.

closed 2020-07-10 by VDelport