Blockstream/satellite #57: blocksat-cli usb launch fails

VDelport, 2020-07-13:
blocksat-cli usb launch

------------------------------- Find DVB adapter ---------------- ---------------
ERROR FE_SET_VOLTAGE: Operation not allowed
Found DVB-S2 adapter: TurboSight TBS 5927

----------------------------- Tuning DVB Receiver ------------------ -----------
Running dvbv5-zap
LNBf L10750 is used
L10750
Frequencies: 11700 to 12200 MHz, LO: 10750 MHz
Demultiplexer »dvb0.demux0« is used.
Channels are read from the file »/home/sdr/.blocksat/config-channel.conf«.
TurboSight TBS 5927 DVB-S / S2 (/ dev / dvb / adapter0 / frontend0) device capabilities:
     CAN_2G_MODULATION
     CAN_FEC_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_QPSK
DVB-API version 5.11, current V5 delivery system: DVBS2
Supported transmission systems:
     DVBS
    [DVBS2]
     DSS
Frequency range for the current standard:
From: 950 MHz
To: 2.15 GHz
Symbol rate change for the current standard:
From: 100 kbauds
To: 70.0 MBauds
Suppose you are in DE

Set to 11484300 Hz
LNBf with a single LO at 0.00 MHz
frequency: 11484.30 MHz, high band: 0
SEC: sets voltage to 13V
DiSEqC SOUND: OFF
DiSEqC SOUND: OFF
L-band frequency: 734.30 MHz (offset = 10750.00 MHz)
ERROR FE_SET_PROPERTY: The argument is invalid
FREQUENCY = 11484300
INVERSION = AUTO
SYMBOL_RATE = 1000000
INNER_FEC = AUTO
MODULATION = QPSK
PILOT = 4294967295
ROLLOFF = AUTO
POLARIZATION = VERTICAL
STREAM_ID = 0
DELIVERY_SYSTEM = DVBS2
ERROR: dvb_fe_set_parms failed
SEC: sets voltage to OFF

What does the error "ERROR FE_SET_VOLTAGE: Operation not allowed" mean?

According to the picture under Hardware Connection, TBS 5927 generates the LNB voltage itself. Is that correct?

blockstreamsatellite, 2020-07-13:
Hi @VDelport ,

You can ignore the error `ERROR FE_SET_VOLTAGE: Operation not allowed`. That's not a problem.

Your problem seems to be on the L-band frequency of 734.30 MHz. This frequency is out of the support range of the TBS5927, which is from 950 to 2150 MHz. Are you reusing the LNB you had on your Blocksat 1.0 setup (presumably with LO frequency of 10750 MHz)? Or are you now using the Universal LNB that comes with the [Basic Kit](https://store.blockstream.com/product/blockstream-satellite-basic-kit/), that is, the UL1-PLL LNB? 

The downlink signal from satellite Telstar 11N Europe is at frequency 11484.3 MHz. Thus, with the 9750.00 MHz LO of a Universal LNB, you will have the L-band frequency at 1734.30 MHz, which is within the tuning range of the TBS5927. In contrast, with a 10750 MHz LNB, you would have the signal at 734.30 MHz (as you seem to have), which is too low. So, in other words, you need an LNB with LO frequency of 9750 MHz, and so Universal LNBs are a good option.

If you don't have a Universal LNB at hand yet, you can look for example for  the GEOSATpro UL1PLL model (which comes on the Basic Kit) or the Avenger PLL321S-2, both of which are [PLL LNBFs](https://github.com/Blockstream/satellite/blob/master/doc/hardware.md#lnb).

Once you change the LNB, don't forget to run the configuration helper (command below) again. This command prompts you for the model (or specifications) of the LNB that you have and uses this information to create a configuration file for the TBS5927. Hence, it's important that you inform the correct frequencies.

```
blocksat-cli cfg
```

VDelport, 2020-07-13:
OK, too bad. Then why is the Maverick MK1-PLL queried during configuration instead of giving a hint that the expensive TSB2927 receiver variant can no longer be used?

blockstreamsatellite, 2020-07-13:
Thanks for the feedback, @VDelport . 

The current version of the configuration helper lists the LNBs that Blockstream Satellite users typically have. The MK1-PLL LNB is still supported, so it is in the list. For example, it can be used with any type of receiver within the Galaxy 18 and Eutelsat 113 regions (Ku high band). Or it can be used with the RTL-SDR receiver in T11N Africa/Europe regions (Ku low band), as you were using.

We will update the CLI to throw an error message when the L-band frequency produced by the LNB is out of the tuning range of the receiver. We will also clarify the TBS5927/S400 tuning range in the documentation, similarly to [how we do for the RTL-SDR](https://github.com/Blockstream/satellite/blob/master/doc/hardware.md#components-for-a-software-defined-radio-sdr-setup).

Apologies for the inconvenience.

blockstreamsatellite, 2020-07-13:
@VDelport FYI, we have just pushed an update to the CLI that takes the LNB frequency range and the receiver tuning range into account during the configuration step.

In this new version (0.2.6), the configuration helper only lists the LNBs that are adequate for the user's region. The suitability of an LNB is judged based on its input frequency range. For example, the MK1-PLL has an input frequency range from 11.7 to 12.2GHz, and, therefore, it is not suitable for the [downlink frequency of 11484.3 MHz](https://github.com/Blockstream/satellite/blob/master/doc/frequency.md#signal-frequencies) within the Telstar 11N Europe region. This is the first LNB requirement listed in the [hardware guide](https://github.com/Blockstream/satellite/blob/master/doc/hardware.md#lnb).

In the specific case of the RTL-SDR, we recognize that you can still try to use the MK1-PLL LNB despite the fact that its rated input frequency range does not cover the frequency of 11484.3 MHz. In this case, you have to select the option "Other" on the LNB configuration section:
```
------------------------------------- LNB -------------------------------------
Please, inform your LNB model:
[ 0] GEOSATpro UL1PLL (Universal Ku band LNBF)
[ 1] Avenger PLL321S-2 (Universal Ku band LNBF)
[ 2] Other
```
Subsequently, you need to provide all the LNB specifications manually, and, in the end, you will be warned about the frequency range, with an option to continue with the out-of-range frequency.
```
WARNING: Your LNB's input frequency range does not cover the frequency of Telstar 11N Europe (11484.3 MHz)
Continue anyway? [N/y]
```

Nevertheless, if you follow this approach to configure the MK1 manually for T11N Europe region while using the TBS 5927, the CLI will check that the resulting [L-band frequency](https://github.com/Blockstream/satellite/blob/master/doc/frequency.md#l-band-frequencies) (in your case of 734.30 MHz) is out of the receiver's tuning range. In this case, the CLI with throw a configuration error, as the L-band must be within the receiver's tuning range.

FYI, the L-band tuning ranges of the supported receivers are the following:
| Receiver | Range | 
|--------- | -------|
| RTL-SDR with R820T2 tuner | 24 to 1766 MHz |  
| RTL-SDR with E4000 tuner | 52 to 2200 |
| TBS 5927 | 950 to 2150 MHz |  
| Novra S400 | 950 to 2150 MHz |  

The frequency range of 950 to 2150 MHz is the typical range of DVB-S2 receivers.

closed 2020-07-15 by VDelport