Blockstream/satellite #58: blocksat-cli sdr cannot synchronize the bitcoin blockchain permanently. Going back to the stable version 1.4.0 would be the best

VDelport, 2020-07-15:
I ran a test with blocksat-cli sdr over 7 hours and came to the conclusion that the Bitcoin blockchain cannot be permanently synchronized with blocksat-cli sdr.

The blockchain had previously been synchronized to 100 percent up to block 639337 via the Internet.

I am sending the terminal output from bitcoind from 2020-07-15.07: 23: 44 to 2020-07-15.13: 29: 14. 
[bitcoind_200715.pdf](https://github.com/Blockstream/satellite/files/4927667/bitcoind_200715.pdf)

blocksat-cli sdr ran on a Mini PC Intel Pentium N4200 (4 x 1.1 GHz / max. 2.5 GHz) 4 GB RAM 64 GB eMMC.

If I have correctly evaluated the output of bitcoind, the result is:

- blocksat-cli sdr received 39 subsequent blocks.
- 19 of 39 blocks were complete, 2 of them after repeated transmission.
- In 36 blocks, only the header was completely detected.
- Not even the header could be detected in one of the 39 blocks.

I think the main problem is that DVB-S2 is being used, which is a technology that is typically used to transmit tv, video and audio where certain losses of the 188-byte TS packets are not a problem for perception . In my opinion, this technology is unsuitable for the transmission of Bitcoin data, where every bit has to be right.

A successful execution of blocksat-cli sdr on a Raspberry PI also appears impossible, since the DVB-S2 reception software leandvb and also tsp require too much CPU performance. If you reduce the CPU performance by the number of ldpc iterations, the probability of faulty TS packets increases.

Conclusion:
Going back to the robust version 1.4.0 with GNU Radio, BPSK and above all with CRC error protection, would be the best. With version 1.4.0 I had measured a total of 0.0 percent packet loss in a similar test over approx. 8 hours and was able to synchronize the blockchain safely, and under exactly the same conditions (satellite antenna, LNB, RTL-SDR, and computer). Implementation on a Raspberry Pi was possible and even ran robustly with the GUI.

blockstreamsatellite, 2020-07-27:
Hi @VDelport , thanks for the comprehensive report of your setup and experiments. I'll split this long answer into sections.

## General Info

We have recently posted a detailed technical explanation regarding how Bitcoin Satellite works. You can find it in the [project's Wiki page](https://github.com/Blockstream/bitcoinsatellite/wiki). I will refer to this document repeatedly in the comments below to facilitate the explanation.

Second general comment is that we have also recently released an update for Bitcoin Satellite. I recommend you upgrade yours to this new release as it may be helpful in various aspects.

Assuming you've installed bitcoin-satellite via `blocksat-cli`, the commands to upgrade are:

First upgrade the CLI:
```
sudo pip3 install blocksat-cli --upgrade
```
Then upgrade all software dependencies, including Bitcoin Satellite:
```
blocksat-cli deps update --btc
```
To confirm the update, you can run `bitcoind --version` after updating. The new version should be like so:
```
Bitcoin Core version v0.19.1.0-g7b38d9fef67f4c05ee719126bccd7c71162e72ee
```

## Experiment duration

> I ran a test with blocksat-cli sdr over 7 hours and came to the conclusion that the Bitcoin blockchain cannot be permanently synchronized with blocksat-cli sdr.

My comment to this is that your observation period is likely too short. I'd recommend reading about [the three streams that are transmitted (multiplexed) over the Blockstream Satellite network](https://github.com/Blockstream/bitcoinsatellite/wiki#satellite-transmissions) and particularly [how the interleaved transmission scheme works](https://github.com/Blockstream/bitcoinsatellite/wiki#interleaved-chunk-transmissions). Using the terminology from the linked article, the important point is that your receiver may fail to receive a block through **Stream 1** at some point. When this happens, it will wait to receive the repetition of the block via **Stream 2**. However, this repetition can take more than 7 hours to complete in its current form. 

When your node loses a block sent via **Stream 1** completely, it needs the full repetition sent through **Stream 2**, hence possibly more than 7 hours (for very long blocks). In contrast, if it loses only a tiny part of the block sent via **Stream 1**, it then only needs a few chunks of data sent through **Stream 2**, so it can recover very quickly. In any case, you can use RPC `getchunkstats` as [described in the Wiki](https://github.com/Blockstream/bitcoinsatellite/wiki#monitoring-the-fec-decoding-process) to get the actual progress of the block in terms of chunks received vs. chunks needed.

## Header Complete but not the Body

> In 19 blocks, only the header was completely detected.
> Not even the header could be detected in one of the 41 blocks.

First, it is worth clarifying the terminology. You can see what are the *header* and *body* data structures [in the Wiki](https://github.com/Blockstream/bitcoinsatellite/wiki#data-structures-used-to-transport-blocks-over-bitcoin-satellite). 

Using the terminology of the linked Wiki page, when you receive a block via **Stream 1**, both the header and the body of the block should complete quickly one after the other. In contrast, when your receiver fails to complete a block via **Stream 1**, as mentioned earlier, your node waits for the repetition via **Stream 2**. In this case, there can be a very long interval from the completion of the header to the completion of the body. This is because the header is a small data structure, whereas the body can be very long. It is also due to the [interleaved chunk transmission scheme](https://github.com/Blockstream/bitcoinsatellite/wiki#interleaved-chunk-transmissions), which promotes significantly better reliability on block transmissions at the expense of reduced block transfer speed. Depending on the block size, this interval could be a few hours and possibly more than 7 hours in case the entire transmission from **Stream 1** was lost at some point.

## Are you receiving Stream 2 at all?

So both comments above assume that you eventually miss a block on **Stream 1** and need its repetition sent via **Stream 2**. With the Linux (TBS 5927) and Standalone (Novra S400) receivers (a.k.a. [Basic and Pro satellite Kits](https://store.blockstream.com/product-category/satellite_kits/)), this always holds, as these devices will always get Stream 1 and 2, and, if the SNR permits, they will also get Stream 3 simultaneously (again, refer to the definition of streams on the [Wiki page](https://github.com/Blockstream/bitcoinsatellite/wiki#satellite-transmissions)).

In contrast, with the SDR receiver, you must choose the [Channel](https://github.com/Blockstream/bitcoinsatellite/wiki#satellite-transmissions). The default SDR channel is the low-throughput **Channel 1**, which only receives Streams 1 and 2. However, if you are running with `blocksat-cli sdr -m high`, you are tuning to **Channel 2** (the high-throughput channel), which only receives Streams 1 and 3. In this case, you would not get the repetitions of the past ~24h of blocks. Instead, you would be getting the stream that is meant for the full chain sync. This would not be a recommended setting for a node that is already in sync if your goal is maximum reliability on staying in sync.

Could you confirm your `blocksat-cli sdr` command and the channel you are tuning to?

## Why you are losing chunks

[... cut ...]
Which RPi model have you used? It is possible that it works on a RPi 4, subject to further testing. Let us know in case you have tested one. It is in our roadmap to work more on RPi compatibility tests.

As discussed in https://github.com/Blockstream/satellite/issues/55#issuecomment-651374837, one can try to reduce the `sdr` options `--ldpc-iterations` and `-n` (number of helper threads) to try to make the SDR receiver work on a RPi. As you pointed out in your comment, there is a tradeoff between their reduction and performance.

VDelport, 2020-07-28:
Thank you for your detailed answer.

Well, let's compare my above test results from blocksat v2.0 with a test I did with blocksat v.1.4 in March 2020. The edition of bitcoind v16 can be found here: 

[blocksat_v1.4_bitcoind_v16_20200311.pdf](https://github.com/Blockstream/satellite/files/4988260/blocksat_v1.4_bitcoind_v16_20200311.pdf)

bitcoind was started by me on March 11th, 2020 at 06:57 (UTC).
At this point, the last block of the blockchain was heigth 621186.

blocksat_rx was only started around 8:30 a.m., about 1.5 hours later.
At this point the current block was height 621190.

bitcoind now continuously receives "old" blocks from height 621133 and in between "new" blocks from height 621193.

At 12:22 p.m. bitcoind receives the block with the height 621187 which it can now append. The progress at this time is = 0.99867.

Among other blocks are blocks 621188 to 621192, which are all successfully appended. All other blocks have already been saved completely by bitcoind and are now successfully added one after the other. While these blocks are being appended, new blocks arrive that are completely saved but cannot yet be appended. In the further course, these blocks are also successfully attached.

Around 2:55 p.m. progress = 0.999998 - the blockchain is synchronized and will remain so until the test is terminated.

You can see that blocksat v1.4 ran absolutely stable. As soon as a block was sent, it could be saved completely.

How long would it take for blocksat v2.0 for the blockchain to reach 0.999998 if you start receiving the blocks 1 to 2 hours later?

The wiki says that Stream 1 allows bitcoin satellite receiver (Rx) nodes to stay in sync with the blockchain in real-time. What do you understand by the term real time? Does realtime mean that it may take more than 7 hours before all missing data chunks of a block are received via Stream 2?

Please leave the issue open until I can report a successful long-term test with blocksat v2.0.

kevinelliott, 2020-12-04:
I have been able to decode the low stream with a Raspberry Pi 4 running Ubuntu 20.04. I have not been able to decode the high stream.

Since it seems the Pi4 is inadequate for decoding the high stream, can you recommend a good (and cheap) mini PC that can be sufficient?

VDelport, 2020-12-05:
@kevinelliott 

Thank you for your information and congratulations. Have you tested your Raspberry Pi 4 (leandvb + tsp) + RTL-SDR receiver over a longer period (days) and were you able to synchronize the blockchain permanently?

As a mini-computer I use an Intel® Compute Card and can also decode the higher-rate data stream with it. The Intel® Compute Card is not exactly cheap, however.

blockstreamsatellite, 2021-05-18:
Hi @VDelport Are you still planning to report results in this ticket? Shall we close it? Thanks.

VDelport, 2021-05-18:
Due to the corona situation, I did not have sufficient access to our satellite system in the last few months, so the planned long-term test is still pending.

blockstreamsatellite, 2021-05-18:
Thanks for your prompt response, @VDelport We will close the issue for the time being. Please feel free to reopen it any time.

closed 2021-05-18 by blockstreamsatellite