Blockstream/satellite #62: Attempt to install blocksat-cli usb without VM fails

VDelport, 2020-08-14:
I'm trying to install blocksat-cli usb without a VM.
I follow their instructions on https://github.com/Blockstream/satellite/blob/master/doc/tbs.md#tbs-5927-drivers.

After blocksat-cli deps tbs-drivers I get:
blocksat-cli dependencies: error: invalid choice: 'tbs-drivers' (choose from 'install', 'update')

After blocksat-cli deps install I get:
Installation of dependencies no supported yet for Linux USB demodulator setup

To install the TBS drivers I ran:
cd util /
./tbsdriver.sh

After sudo blocksat-cli usb config I get:
 Find DVB Adapter -------------------------------
Traceback (most recent call last):
  File "/ usr / local / bin / blocksat-cli", line 11, in <module>
    sys.exit (main ())
  File "/usr/local/lib/python3.6/dist-packages/blocksatcli/main.py", line 60, in main
    args.func (args)
  File "/usr/local/lib/python3.6/dist-packages/blocksatcli/usb.py", line 566, in usb_config
    common_params = _common (args)
  File "/usr/local/lib/python3.6/dist-packages/blocksatcli/usb.py", line 551, in _common
    adapter, frontend = _find_adapter ()
  File "/usr/local/lib/python3.6/dist-packages/blocksatcli/usb.py", line 97, in _find_adapter
    res = subprocess.call (cmd, stdout = devnull, stderr = devnull)
  File "/usr/lib/python3.6/subprocess.py", line 287, in call
    with Popen (* popenargs, ** kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type (errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'dvbnet': 'dvbnet'

blockstreamsatellite, 2020-08-14:
Hi @VDelport,

It looks like you are using an old version of the CLI. Could you check `blocksat-cli -v`? The latest version is `0.2.9`.

Please upgrade by running:
```
sudo pip3 install blocksat-cli --upgrade
```

VDelport, 2020-08-14:
Thank you. How about the hint
sudo pip3 install blocksat-cli --upgrade on the page https://github.com/Blockstream/satellite/blob/master/doc/tbs.md#tbs-5927-drivers to be added?

blockstreamsatellite, 2020-08-14:
Thanks for your suggestion @VDelport Added: https://github.com/Blockstream/satellite/blob/master/doc/tbs.md#tbs-5927-drivers

closed 2020-08-14 by VDelport