Blockstream/satellite #71: Compressed Transactions

VDelport, 2020-10-04:
First of all, sorry for asking this question in the issues. I received blocks with blocksat-cli via channel 2 and saved them in a file. The received file has a size of approx. 4 Mbytes. The file contains a number of completely received blocks with a total size of approx. 4.8 Mbytes. The wiki https://github.com/Blockstream/bitcoinsatellite/wiki#compressed-transactions states that the data of the transactions are compressed before they are transmitted. Is this the reason for the difference in the amount of data?

blockstreamsatellite, 2020-10-09:
Hi @VDelport ,

Thanks for your question. That's right, the FEC-encoded version of the block, which is sent over satellite, carries compressed transactions. However, could you clarify what is the "difference in the amount of data" you are referring to? 

Note that the block size varies significantly per block and over the blockchain. The blocks at the beginning of the blockchain are very small, whereas more recent blocks are larger. So your result depends on the blocks you received. As explained on [the Wiki you mentioned](https://github.com/Blockstream/bitcoinsatellite/wiki#interleaved-chunk-transmissions), channel 2 iterates over the full chain, but with a transmission window of 4000 blocks. At any point in time, it is transmitting 4000 blocks in parallel. Once these blocks complete, the transmission window advances. So, for example, if you collected the file when channel 2's transmission window was at the beginning of the blockchain, you could have many blocks in less than 4 MB.

When did you run this experiment? Channel 2 was around height 490k on T11N EU 5 days ago (when you asked).

VDelport, 2020-10-10:
Thank you for your response. I recorded the TS file on June 26th, 2020. The blocks that were recognized as complete in the TS file by bitcoinsatellite date from 07/11/2015 and had an average block size of approx. 230 bytes.

closed 2020-11-17 by VDelport