Blockstream/satellite #72: Can’t sync to tip with txindex=1/reindex during IBD

bg002h, 2021-01-11:
Not sure if this is an important issue or not, but, if I start syncing via satellite from scratch with transaction index turned on, sync fails with some Error about ReadBlockFromDisk: Deserialize or I/O Error - ReadCompactSize(): size too large: iostream…I have replicated this issue on two different drives. Reindexing seems to trigger the problem. 

Perhaps index/reindex should not happen during IBD? 

I can post a more formal issue with debug log if there is any interest in this issue. I’ve just been playing around to be certain I can replicate the issue…

bg002h, 2021-03-04:
As best I can tell, nobody running this fork can successfully -reindex (txindex=1 or 0). Give it a try and see if you can. It might be possible if wiping future blocks/partial blocks and/or disabling udpmulticast stuff.

blockstreamsatellite, 2021-03-04:
Hi @bg002h ,

Thanks for the detailed description. We will investigate the issue and get back to you soon.

blockstreamsatellite, 2021-04-07:
Hi @bg002h ,

We are currently investigating this problem. Do your recall, or could you confirm whether the reindexing always fails on the same block? In other words, if you try to reindex and it fails, what happens when you try again? Does it fail on the same block (same `FlatFilePos`)? 

Another useful test is like so:

1. Create a new Bitcoin data directory (datadir) by copying from an existing datadir using `rsync`.
```
rsync -rah old_datadir/ new_datadir/
```
2. Try to reindex on the new datadir.
3. If the reindexing fails, run `rsync` again as explained in step 1, and then proceed with step 2. See if the reindexing fails on the same block.

It seems more likely that the failure will happen randomly and not on the same block. That would indicate random disk corruption.

There are a number of threads reporting this error (`ReadBlockFromDisk : Deserialize or I/O error - ReadCompactSize() : size too large`). For instance, https://github.com/bitcoin/bitcoin/issues/5668, https://github.com/bitcoin/bitcoin/issues/14999, and https://github.com/bitcoin/bitcoin/issues/17144. The discussions indicate the problem is likely disk corruption.

Lastly, could you check what kind of disk and filesystem is holding your bitcoin datadir? More specifically, the output from:

```
sudo fdisk -l
```
and
```
lsblk -f
```

If you are not sure which filesystem is holding the bitcoin datadir, you can check the output of `df` (or see the output of `lsblk -f` directly).

bg002h, 2021-04-07:
>    1. Try to reindex on the new datadir.
>    2. If the reindexing fails, run rsync again as explained in step 1,
> instance, bitcoin/bitcoin#5668
> <https://github.com/bitcoin/bitcoin/issues/5668>, bitcoin/bitcoin#14999
> bitcoin/bitcoin#17144 <https://github.com/bitcoin/bitcoin/issues/17144>.

Random blocks. I’ve tried multiple disks with similar results. Waiting for
node to sync to tip and then indexing works fine.

I’ve been using ext4 on NVME on a PCIE. I’ll try messing around some more
this week or next.

blockstreamsatellite, 2021-05-18:
@bg002h In case you find a chance to investigate the issue further, you can try to reproduce it with the latest Bitcoin Satellite version based on Bitcoin Core 0.21. 

You can update the Bitcoin Satellite package using the CLI with the command `blocksat-cli deps update --btc` (see [here](https://blockstream.github.io/satellite/doc/quick-reference.html#bitcoin-satellite-setup)) or directly through your package manager (apt/dnf/yum).

bg002h, 2021-05-20:
Awesome. Will give it a whirl, if I’m not already running it. I just got my
basestation setup last week (man that was a lot easier than installing a
dish!).

lomonosofff, 2021-12-14:
А что мне делать если у меня виндус и  при попытке отправить биткоин он висит не попадая в мемпул а в логах ошибки вот такие же как вы обсуждаете! Такая перда происходит с мая месяца и я уже и диски новые купил и реиндексы всякие проделал а главное перенести по приватному ключю не могу, оно в облаках разворачивается пустое. Как это и что делать?

asuka431, 2022-02-27:
i solved this kind of probrem when i checked permissions of blocks.
for my time it wasn't writable

closed 2023-03-03 by blockstreamsatellite