Blockstream/satellite #76: R82XX No valid PLL values for 3958282424 Hz!

santimaster2000, 2021-06-09:
I'm trying to sep up my sat receiver, using this SDR:

https://www.amazon.com/gp/product/B01HA642SW/ref=as_li_qf_sp_asin_il_tl?ie=UTF8&tag=cryptohardw09-20&camp=1789&creative=9325&linkCode=as2&creativeASIN=B01HA642SW&linkId=c726873c6574c583d70fab3ff2de71f2

I'm from South America (Argentina), trying to connect to Eutelsat 113, my LNB LO is 10750, my LNB is a Maverick MK1-PLL, while trying to calculate the right frequency:

https://github.com/Blockstream/satellite/blob/master/doc/frequency.md

I'm coming up with 1.316,9 Mhz (12066.9 MHz - 10750 MHz), problem is, when I run the command "blocksat-cli --gui --derotate 1316900000 -d" I get this error:

https://i.imgur.com/O9bK7db.jpg

Not sure if I'm doing something wrong or if this particular SDR is not compatible with my local satellite.

Thanks in advance and sorry for my bad english, not my native language but I try :)

VDelport, 2021-06-09:
Hello,

the RTL-SDR receiver has an R820T2 tuner, and this tuner should be able to process frequencies up to approx. 1700 MHz.

The question is, how does the output "No valid PLL values for 3958282424 Hz" come about? Does the specification of the frequency after the derotate option in the command line correspond to the documentation?

blockstreamsatellite, 2021-06-09:
Hi @santimaster2000 ,

Thanks for getting in touch. The problem is on the command that you are executing. The `--derotate` parameter is only meant to correct a small frequency offset, usually less than +-1 MHz. It is not supposed to take the L-band frequency in the GHz range (1.3169 GHz in your case). The latter, the L-band frequency, is computed automatically by the CLI, so you don't need to compute it manually (although you did compute it correctly).

The `--derotate` parameter is just a convenient way to account for the fact that your LNB 's local oscillator (LO) will not produce a perfect 10750 MHz frequency. In practice, the LO frequency presents an offset relative to the nominal value, which can shift the signal out of the RTL-SDR's reception range if not accounted for.

You can usually find a reasonable fixed value for the `--derotate` parameter when aligning the antenna for the first time. Please follow the instructions in [the antenna pointing guide](https://blockstream.github.io/satellite/doc/antenna-pointing.html#pointing-with-an-sdr-based-receiver). If you are unsure about the frequency offset to derotate, just omit this parameter for now and try running like so:

```
blocksat-cli sdr --gui
```

Also, the components you mentioned are all good. The RTL-SDR with the R820T2 tuner and the Maverick MK1-PLL LNB are known to work well for reception from Eutelsat 113.

santimaster2000, 2021-06-10:
> Hi @santimaster2000 ,
>
> Thanks for getting in touch. The problem is on the command that you are executing. The `--derotate` parameter is only meant to correct a small frequency offset, usually less [...]

Thank you so much ! I get it know, I know this is not totally related to the main issue, but, any advice on how to align the dish ? I live on an 8th floor and my phone's GPS/Accelerometer are not very good.

blockstreamsatellite, 2021-06-10:
@santimaster2000 Do you have line-of-sight to the satellite area in the sky? That is, no obstacles such as other buildings in front?

Do you have the pointing angles (azimuth, elevation, and polarity) for your location already? If not, you can get them by using the [coverage map](https://blockstream.com/satellite/#satellite_network-coverage). Once you have them, you can follow the instructions in the [antenna mounting section on the user guide](https://blockstream.github.io/satellite/doc/antenna-pointing.html#mount-the-antenna).

If your phone's GPS/accelerometer is not providing accurate readings, you can use it only for a rough indication of where the satellite is. That should help already. After that, you will need a bit of trial and error until you can visualize the signal on GQRX or the CLI in GUI mode (with command `blocksat-cli sdr --gui`), as instructed on [the user guide](https://blockstream.github.io/satellite/doc/antenna-pointing.html#pointing-with-an-sdr-based-receiver).

Aside from that, there are a couple of things to consider when pointing with an SDR receiver. First, what kind of antenna are you using? A regular satellite dish? If it is a dish, do you know if it's an [offset dish](https://en.wikipedia.org/wiki/Offset_dish_antenna)? That's the most typical dish type in Ku band, and it has slightly different pointing instructions in terms of elevation. The elevation angle you get from the [coverage map](https://blockstream.com/satellite/#satellite_network-coverage) is the nominal elevation used by a *prime-focus dish*. In contrast, an *offset dish* requires an extra tilt in elevation corresponding to the dish's offset angle (specified by the manufacturer). 

For instance, if you are in Buenos Aires, the nominal elevation angle will be around 20º. Meanwhile, if you were using an offset dish with an offset angle of 21º (see [this dish](https://www.triax.com/products/satellite/satellite-dishes/aluminium-dishes/ae-75-quatro-switch-ral-7035-kit), for example), the required elevation would be -1º (i.e., the subtraction `20 - 21`) instead of the nominal 20º. With such a negative elevation, it would look like the dish was pointing down. However, in reality, that's just the required elevation tilt to make the reflections converge to the LNB.

The other aspect to keep in mind with an SDR receiver concerns the polarization angle (also referred to as *polarity angle*). The issue that may arise depends on the LNB power inserter. More specifically, the power inserter may force the LNB into a polarization mode other than the required one. 

For example, you want to receive from E113, which is vertically polarized. However, if you are using the DirecTV/AT&T 21V SWM power inserter (a commonly used model), the inserter will supply 21 Volts to the LNB, configuring it with horizontal polarization (refer to [the SDR hardware section](https://blockstream.github.io/satellite/doc/hardware.html#software-defined-radio-sdr-setup)). In this case, the polarity angle obtained by the [coverage map](https://blockstream.com/satellite/#satellite_network-coverage) will not work. You will need to rotate the LNB by an extra 90º in the most convenient direction (usually to stay within +-90º). For example, in Buenos Aires, the nominal polarization angle might be around -49º. In this case, if you are using a 21V SWM power inserter, you can rotate the LNB to a polarization angle of +41º (i.e., `-49 + 90`).

santimaster2000, 2021-06-10:
> @santimaster2000 Do you have line-of-sight to the satellite area in the sky? That is, no obstacles such as other buildings in front?
>
> Do you have the pointing angles (azimuth, elevation, and [...]

OK, I'm kinda moving forward with this, this is my ghetto style dish/base:

https://i.imgur.com/5a6qeRN.jpg

It's a DirecTV dish, measured up and down it's 26 inches, measured from side to side it measures 23.5 inches, thing is, this is my "elevation control":

https://imgur.com/a/6BaUWpV

So, if I set it anywhere close to 20º, it's basically pointing down.

Any ideas ?

santimaster2000, 2021-06-12:
OK, this is my "line of sight", E113 should be above the pyramid top looking building and the one with the HSBC logo, above, in the middle:

https://i.imgur.com/JUgvcl3.jpg

And this is the "gap" in my balcony, 41 inches from the floor, that's glass, guessing signal can go trough it, above, the "window" where it's actually open, it's 45 inches, from the railing to the ceiling:

https://i.imgur.com/7SAYUUu.jpg

Think I have a good enough angle ? I can bring the dish back into my apartment and just leave the window open if I have to in order to get enough clearance.

blockstreamsatellite, 2021-07-27:
HI @santimaster2000 ,

Apologies for the delay. Did you manage to have success pointing to E113?

Regarding your comment:

> So, if I set it anywhere close to 20º, it's basically pointing down.

That's right. You are using an offset dish, and the offset angle is probably greater than 20º. Hence, for an elevation of 20º, it will appear as if the dish were pointing down. That's the correct configuration.

> And this is the "gap" in my balcony, 41 inches from the floor, that's glass, guessing signal can go through it, above, the "window" where it's actually open, it's 45 inches, from the railing to the ceiling:

The glass will likely introduce significant attenuation and make the signal too weak to be detected. If possible, try pointing with the window open first. Then, if successful, you can try closing the window to get a sense of how much attenuation the glass introduces. 

The window space on the picture seems sufficient if you get your dish close enough to it. Ideally, the dish would be completely out of the window to avoid any undesirable indoor reflections. However, I assume that's not a possibility in your case. So try to bring the dish as close as possible to the window and make sure the dish is entirely exposed to the sky.

santimaster2000, 2021-07-30:
> HI @santimaster2000 ,
>
> Apologies for the delay. Did you manage to have success pointing to E113?
>
> Regarding your comment:
>
>
> That's right. You are using an offset dish, and the offset angle is [...]

No reason to apology, I 100 % understand I'm basically getting free tech support for a DIY project, no, to be honest, haven't been trying lately because I don't have any effective way of measuring if I'm getting any closer, was considering buying some kind of signal measuring device, like the Freesat v7 hd and see if that helps, what do you think ?

Again, thank you so much for the help :)

blockstreamsatellite, 2021-08-03:
You are welcome, @santimaster2000 

> was considering buying some kind of signal measuring device, like the Freesat v7 hd and see if that helps, what do you think ?

Not sure about the device you mentioned, but a satellite finder can be helpful when pointing. You can find some instructions for pointing with a satellite finder on the [antenna pointing guide](https://blockstream.github.io/satellite/doc/antenna-pointing.html#pointing-with-a-satellite-finder).

Also, whenever you try pointing again, feel free to contact us for real-time support. You can join the [Telegram group](https://t.me/blockstream_satellite_feed) or contact us at the **#blockstream-satellite** IRC channel on freenode.

santimaster2000, 2021-08-04:
> You are welcome, @santimaster2000
>
>
> Not sure about the device you mentioned, but a satellite finder can be helpful when pointing. You can find some instructions for pointing with a satellite finder [...]

Thanks again, yeah, I'm gonna join the Telegram group, I'm planning on putting on some work this weekend.

Camille8383, 2022-06-13:
> I'm trying to sep up my sat receiver, using this SDR: [...]

santimaster2000, 2022-06-13:
> I'm trying to sep up my sat receiver, using this SDR: [...]

Thanks again, in the end, I didn't have proper line of sight, that's why I was unable to lock on a signal.

closed 2023-03-03 by blockstreamsatellite