Blockstream/satellite #77: Any chance of ever having satelite software detached from Bitcoin Core?

Kixunil, 2021-07-17:
I love the idea of Bitcoin satellite but I'm not very happy with having to run a fork of Bitcoin Core. I see these disadvantages:

* Slower updates - satellite fork seems to be behind Bitcoin Core
* Decreased security - Bitcoin Core is one of the most reviewed software, much more than any fork. If satellite was a detached service that only supplies the blocks to Core over P2P interface it could be treated as untrusted, sandboxed, and together with cross-verification of headers made very safe. Same could be done by running both upstream Core and the fork but that would have a huge impact on performance. Core itself is already very resource consuming, doubling it is not great
* Compatibility issues - I'd like to package this software in [Cryptoanarchy Debian Repository](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder), with current architecture I'd have to setup replacing of the default `bitcoind` package and worry about interactions between the versions, including all other applications. Having the service detached would make me only worry about two services communicating (over P2P, which is a stable interface).

Is there any hope the two will one day become separated? Is there any strong reason not to that I'm not seeing?

blockstreamsatellite, 2021-07-26:
Hi @Kixunil ,

Thanks for sending your thoughts on Bitcoin Satellite. Here is some extra information that you and others may find helpful.

> Slower updates - satellite fork seems to be behind Bitcoin Core

Indeed, Bitcoin Satellite is not updated synchronously with Bitcoin Core. Hence, for some time, it may fall behind Core. However, note that, at this point, the latest Bitcoin Satellite version is based on Bitcoin Core v0.21.1. You can find more information on the [release notes](https://github.com/Blockstream/bitcoinsatellite/blob/0.21.1/doc/release-notes/satellite/release-notes-0.2.1.md), including how to install the package via `apt` from our [Launchpad repository](https://launchpad.net/~blockstream/+archive/ubuntu/satellite/+packages) or via `dnf` from our [Copr repository](https://copr.fedorainfracloud.org/coprs/blockstream/satellite/packages/).

Also, on Bitcoin Satellite's Github repository, you can find the source code for all versions that are currently available:
https://github.com/Blockstream/bitcoinsatellite/branches.

> If satellite was a detached service that only supplies the blocks to Core over P2P interface it could be treated as untrusted, sandboxed, and together with cross-verification of headers made very [...]

We understand your point. For your information, note Bitcoin Satellite can be used both in standalone mode or in conjunction with a Bitcoin Core node. In both cases, however, Bitcoin Satellite runs a full node on its own.

For those concerned with running less widely tested software, the combination of Bitcoin Satellite and Core can be an attractive alternative. For example, you can run Bitcoin Core as the primary node on your preferred host and Bitcoin Satellite as an auxiliary node on a separate physical or virtual host. Then, you can keep the two connected as regular Bitcoin peers. Refer to the [Blockstream Help Center article about this use case](https://help.blockstream.com/hc/en-us/articles/900002374566-Can-I-connect-an-official-Bitcoin-Core-client-to-Blockstream-Satellite-).

You are correct that this would require running two `bitcoind` instances, which can be resource-consuming. Unfortunately, however, it's not currently possible to run Bitcoin Satellite as a fully detached service, in the sense that you suggested. The main disadvantage of implementing Bitcoin Satellite as a fully independent service would be that it would always require communication with a Bitcoin Core node. Hence, it would not be possible to run it standalone.

For instance, one of the key strategies to speed up the download of blocks over satellite is the idea of filling up the block data in advance based on mempool transactions (see the [wiki](https://github.com/Blockstream/bitcoinsatellite/wiki#how-bitcoin-satellite-improves-block-transmission-speeds)). That requires a fully functional mempool. Hence, it is one out of many examples of Core usage by Bitcoin Satellite. 

The other factor to consider is how this project evolved. It originated as a fork of [Bitcoin FIBRE](https://bitcoinfibre.org), which itself was designed to operate as a full node rather than a detached service.

Nevertheless, it is certainly not impossible to implement a fully detached version of Bitcoin Satellite. It only requires significant rework due to how the implementation leverages the Bitcoin Core infrastructure. If you are curious and familiar with the Core source code, here is the last line of code before feeding a block received over satellite into the Core block processing logic: https://github.com/Blockstream/bitcoinsatellite/blob/0.21.1/src/udprelay.cpp#L955 . For instance, one could attempt to redirect this data to a separate full node instead of processing it locally.

>  with current architecture I'd have to setup replacing of the default bitcoind package and worry about interactions between the versions, 

Indeed, the Core and Satellite packages provide applications with the same name (bitcoind, bitcoin-qt, bitcoin-tx, etc.). These will conflict if both packages are installed on the same host. For packaging, it may be helpful to consider using a [program prefix](https://www.gnu.org/software/automake/manual/html_node/Renaming.html) on autoconf's configure step. Alternatively, at installation time, it may be possible to use  `dpkg-divert`. 

As a side note, it's worth mentioning that Bitcoin Satellite can be easily executed on a Docker container. The [`blocksat-host`](https://hub.docker.com/r/blockstream/blocksat-host) image available on Docker Hub has Bitcoin Satellite already installed.

Please feel free to send further questions or comments.

Kixunil, 2021-07-27:
Thank you for in-depth explanation! I don't think I understand what you mean by this:

> The main disadvantage of implementing Bitcoin Satellite as a fully independent service would be that it would always require communication with a Bitcoin Core node.

Do you mean that you'd no longer be able to provide standalone version or do you mean that cutting connection even for a short time would break it? Or perhaps something else?

In case you want to provide both, the maim logic could be separated out (as a "library") and then having short piece of code in Core and a simple daemon should be minimum effort.

If connection breaking is a problem isn't it same problem as power outage? The software doesn't operate for a while. The satellite daemon could just suspend all activity in that case. Bitcoin Core could fall back to getting blocks from the network.

Note that configuring connections is not a problem. My repository is so good at configuring stuff that I had to put a [warning for users not to try](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder#how-to-setup-the-beta-debian-repository). :)

I probably won't be able to rewrite it myself anytime soon but hopefully one day. Thanks for pointers anyway!

ekzyis, 2022-07-10:
> Do you mean that you'd no longer be able to provide standalone version or do you mean that cutting connection even for a short time would break it? Or perhaps something else?

I think he meant that providing a fully independent version would not be possible to run since it _needs_ a Bitcoin node (usage of mempool given as an example) hence the decision to include a Bitcoin node. At least that's how I understood it.

Kixunil, 2022-07-10:
Yes, and I don't see why *communication* would be a problem. Harder to setup, yes, but with benefits that seem worthwhile to me. I also think it should be possible to have both options (Clean Architecture).

blockstreamsatellite, 2023-03-03:
Hi @Kixunil @ekzyis ,

We will close this issue for now, as it has been open and inactive for some time. The points raised here are still valid. It would be nice to have a detached version of Bitcoin Satellite as an alternative to the current fully self-contained application, including the Bitcoin Core functionality. The idea is certainly feasible, with significant rework. However, for now, we recommend the alternatives discussed above, such as running Bitcoin Satellite with satellite connection only and another local Bitcoin Core instance connected to the satellite instance, despite the resource utilization that this setup requires (two bitcoind instances).

Regards

closed 2023-03-03 by blockstreamsatellite