Blockstream/satellite #81: blocksat-cli sat-ip -- simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

SeverinAlexB, 2021-09-22:
Hey guys

I just installed the Blockstream satellite and ran `blocksat-cli sat-ip` and I received an error:

```bash
$ blocksat-cli sat-ip
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/azhodl/.local/lib/python3.7/site-packages/blocksatcli/satip.py", line 608, in _monitoring_thread
    status = sat_ip.fe_stats()
  File "/home/azhodl/.local/lib/python3.7/site-packages/blocksatcli/satip.py", line 512, in fe_stats
    info = rv.json()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

Antenna: SelfSat IP22 Sat>IP Antenna (566 × 300 × 81 mm)

Thanks for your help!

blockstreamsatellite, 2021-09-22:
Hi @SeverinAlexB ,

Thanks for getting in touch. Could you run the command in debug mode and share the results:

```
blocksat-cli --debug sat-ip
```

SeverinAlexB, 2021-09-22:
Here we go:
```bash
2021-09-22 19:54:44 DEBUG    root [Debug Mode]
2021-09-22 19:54:44 DEBUG    blocksatcli.update Last update check was less than a day ago: 0:55:09.075236
2021-09-22 19:54:47 DEBUG    urllib3.connectionpool Starting new HTTP connection (1): 192.168.1.226:8000
2021-09-22 19:54:47 DEBUG    urllib3.connectionpool http://192.168.1.226:8000 "GET / HTTP/1.1" 200 481
2021-09-22 19:54:47 DEBUG    blocksatcli.satip Local Sat-IP client address: 192.168.1.8
2021-09-22 19:54:47 DEBUG    urllib3.connectionpool Starting new HTTP connection (1): 192.168.1.226:8000
2021-09-22 19:54:47 DEBUG    urllib3.connectionpool http://192.168.1.226:8000 "GET /gmi_sw_ver.txt HTTP/1.1" 200 7
2021-09-22 19:54:47 DEBUG    urllib3.connectionpool Starting new HTTP connection (1): 192.168.1.226:8000
2021-09-22 19:54:47 DEBUG    urllib3.connectionpool http://192.168.1.226:8000 "POST /cgi-bin/login.cgi HTTP/1.1" 200 None
2021-09-22 19:54:47 DEBUG    blocksatcli.satip Run:
2021-09-22 19:54:47 DEBUG    blocksatcli.satip tsp --realtime --buffer-size-mb 1.0 --max-flushed-packets 10 --max-input-packets 10 -I http http://192.168.1.226/?src=1&freq=12016.4&pol=h&ro=0.2&msys=dvbs2&mtype=qpsk&plts=on&sr=1000&fec=35&pids=32 --infinite -P mpe --pid 32 --udp-forward --local-address 127.0.0.1 -O drop
2021-09-22 19:54:48 DEBUG    urllib3.connectionpool Resetting dropped connection: 192.168.1.226
2021-09-22 19:54:48 DEBUG    urllib3.connectionpool http://192.168.1.226:8000 "GET /cgi-bin/index.cgi?cmd=frontend_info HTTP/1.1" 200 None
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/azhodl/.local/lib/python3.7/site-packages/blocksatcli/satip.py", line 608, in _monitoring_thread
    status = sat_ip.fe_stats()
  File "/home/azhodl/.local/lib/python3.7/site-packages/blocksatcli/satip.py", line 512, in fe_stats
    info = rv.json()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

`192.168.1.226` is the correct IP.

blockstreamsatellite, 2021-09-22:
Thanks, that was helpful.

Just to double-check. What is the result of the following command?

```
curl http://192.168.1.226:8000/gmi_sw_ver.txt
```

Secondly, there is one way in which the CLI probably won't crash. If you run it with option `--no-fe-monitoring`, like so:

```
blocksat-cli sat-ip --no-fe-monitoring
```

However, that is not very useful when you still need to point the antenna. Without monitoring the frontend, you won't be able to tell whether the antenna/receiver is locked to the Blockstream Satellite signal. 

Third, what happens when you try to access the antenna's web interface on the browser at `http://192.168.1.226:8000`? Do you see the login page?

azhodl, 2021-09-22:
Hello, this is actually my project that Sevi is helping me with. The result of the first command is 3.1.18

The <--no-fe-monitoring> produces no results. Seems to be processing but nothing is happening.

I am able to view the login screen!

Thank you.

blockstreamsatellite, 2021-09-22:
Thanks. That is correct, with the `--no-fe-monitoring` option, the CLI doesn't print any output. This option is just an alternative in case you don't want the logs, and would rather monitor the frontend by accessing the web interface (see the info on the "Frontend Information" tab). 

The next thing I suggest is the following. In one terminal window, keep the CLI running without any frontend monitoring. Run the command you used earlier:

```
blocksat-cli sat-ip --no-fe-monitoring
```

On another terminal window, try to request the frontend information manually using cURL, like so:

- First try a login:

```
curl -X POST -d "username=admin&password=admin&cmd=login" --cookie-jar /tmp/satip-login.cookie http://192.168.1.226:8000/cgi-bin/login.cgi
```

If the above is successful, it will print some HTML content and save the cookie file at `/tmp/satip-login.cookie`. Next, request the frontend info:

```
curl --cookie /tmp/satip-login.cookie http://192.168.1.226:8000/cgi-bin/index.cgi?cmd=frontend_info 
```

Let me know what you see as the frontend information. You could even try to parse it as JSON, which seems to be the step that is failing. To do so, pipe the output to `python3 -m json.tool`, like so:

```
curl --cookie /tmp/satip-login.cookie http://192.168.1.226:8000/cgi-bin/index.cgi?cmd=frontend_info | python3 -m json.tool
```

azhodl, 2021-09-22:
I do not have login credentials yet. Did I skip over this step during setup? 

Below are the commands while the CLI is running in the other terminal:

$ curl -X POST -d "username=admin&password=admin&cmd=login" --cookie-jar /tmp/satip-login.cookie http://192.168.1.226:8000/cgi-bin/login.cgi
<html>

<head>
	<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
	<link type="text/css" href="/webui/styles/global.css" rel="stylesheet" />
	<script type="text/javascript" src="/webui/js/jquery-1.10.2.min.js"></script>
	<script type="text/javascript" src="/webui/js/global.js"></script>
	<script type="text/javascript" src="/webui/js/error.js"></script>
	<title>SAT>IP Antenna - Error</title>
</head>

<body>
<h1>Error</h1>

<p>Something goes wrong ...</p>
<p>If you just tried to login, you probably entered wrong username/passwrod combination.
<br>You can try to login again <a href="/">login page</a>.
<br>OR you will be automiatically redirected in <label id="count_down_val"></label> seconds.</p>

</body>

And the second request:

curl --cookie /tmp/satip-login.cookie http://192.168.1.226:8000/cgi-bin/index.cgi?cmd=frontend_info 
<html>

<head>
	<title>GSS4P SAT>IP Antenna - Login</title>
	<link href="/webui/styles/login.css" type="text/css" rel="stylesheet" />
	<script type="text/javascript" src="/webui/js/jquery-1.10.2.min.js"></script>
	<script type="text/javascript" src="/webui/js/login.js"></script>
</head>

<body>
  <div class="framecontent">
    <div id="heading">
	GSS4P [image] Antenna - Login
    </div>
	<br>
	<br>
	<form action="/cgi-bin/login.cgi" method="POST">

		<div>
			<input type="hidden" name="cmd" id="cmd" value="login" />
		</div>
		<label>Username:</label>
		<div>
			<input type="text" name="username" id="username" />
		</div>
		<br>
		<label>Password:</label>
		<div>
			<input type="password" name="password" id="password" />
		</div>
		<br>
		<div>
			<input type="submit" value="Login">
		</div>

	</form>
  </div>
</body>
</html>

blockstreamsatellite, 2021-09-22:
Ok. Thanks for the feedback. Did you change the default username and password by any chance? The default credentials are:

Username: admin
Password: admin

azhodl, 2021-09-22:
I tried those credentials and they do not work. I did not reach a point in setup where is asked me to set a login/password.

blockstreamsatellite, 2021-09-22:
Right. In that case, you can try a factory reset by running the following request with cURL:

```
curl -X POST -d "cmd=gmim_fact_default" http://192.168.1.226:8000/cgi-bin/gmi.cgi
```

azhodl, 2021-09-22:
I think that worked! I am able to login and now when I call < blocksat-cli sat-ip>

2021-09-22 22:08:22  Lock = False; Level = -71.67dBm; Signal Quality = 0.0%;

I will get the alignment taken care of. I really appreciate the fast and thorough response!!

blockstreamsatellite, 2021-09-22:
Thanks for the feedback. You are very welcome 👍 . Glad it worked! 

Please feel free to get in touch if you have more questions along the process. We'll close this issue.

closed 2021-09-22 by blockstreamsatellite

blockstreamsatellite, 2022-03-08:
For other users accessing this thread in the future, here is another solution to perform a factory reset:

- First, access the Selfsat IP22 over ssh. Username `root`, password `SELFSAT`. 
- Next, run the following command:

```
/opt/bin/gmi_default.sh
```

Then, wait until it reboots.