Blockstream/satellite #83: How must blocksat be configured in order to receive streams 2 and 3 with approx. 70 kbps?

VDelport, 2021-10-14:
Hi there,

In the wiki about Blockstream Satellite it says: "Streams 2 and 3 operate with approximately 70 kbps each. For Stream 2, this bitrate is enough to cycle over 144 blocks at least three times during 24h".

I am receiving blockstream satellite with a TBS5927. 

What has to be entered in the configuration file config-channel.conf so that streams 2 and 3 are really only received with the specified 70 kbps. The data rate that I am currently measuring is approx. 800 kbps.

VDelport, 2021-10-17:
Thanks for the detailed answer.

>However, before that, how are you setting up the transmission? How are you feeding data to the transmitter in the terrestrial transmission system you mentioned?

I use the command option with which you can write the transport stream to a file. Instead of the file, I write the transport stream in a pipe. The pipe is passed to netcat, which forwards the transport stream to the transmitter of my radio transmission system via UDP. The receiver of my radio transmission system transfers the transport stream to tsp via a pipe, which then transfers the blockchain data to bitcoind.

>We could make a branch with the referred change for you to test. 

That would be great. 

>So you could simply experiment with the Tx options on your terrestrial system.

That sounds interesting. If I understand correctly, can I generate a transport stream myself with embedded blockchain data that is already available on the synchronized node in the .bitcoin directory?

It would be good if these possibilities were also documented. :)

VDelport, 2021-10-29:
I created bitcoind from the sources and it works when I set the send_rep_blks = false option. :)
 
How does the system behave if you only set relay_new_blks = true, i.e. send_rep_blks = false? Are the blocks of the last 24 hours then sent or is only the block sent that was just attached to the blockchain as a new block?

The last block of my blockchain currently has the height 707274 and was mined 6 hours ago, i.e. within the last 24 hours. But I find out via Wireshark that no bitcoin data is being sent.

blockstreamsatellite, 2021-10-29:
Good that it works now.

If you use the following tx configuration:

```
ifname=lo
dest_addr=239.0.0.2:4434
ttl=3
bw=80000
txn_per_sec=10
send_rep_blks=true
relay_new_blks=true
```

It will:

1. Try to send **up to** 10 mempool transactions per second. You can check that with the following RPC:
```
bitcoin-cli gettxntxinfo
```

2. Relay new blocks as soon as they arise (`relay_new_blks=true`). You can check if new blocks were sent already by checking the internal transmission buffers.
```
bitcoin-cli gettxqueueinfo
```

Buffers 0 and 1 are dedicated to new blocks. Buffer 2 is dedicated to transactions. Lastly, buffer 3 is dedicated to historic blocks. So you can verify what is being transmitted if you monitor these buffers with something like:

```
watch -n1 -d "bitcoin-cli gettxqueueinfo"
```

3. Send historic blocks (`send_rep_blks=true`). If you don't configure a `depth` option, the default will be to send the entire blockchain (same as `depth=0`). You can check which historic block is being transmitted by running:

```
bitcoin-cli gettxwindowinfo
```

Or a more specific query for Tx stream 0-0:

```
bitcoin-cli gettxwindowinfo 0 0
```

If you want to send the past 24h of blocks, you need to add `depth=144` to your `tx.conf` (there are 144 blocks in 24h on average).

Lastly, bear in mind that 80 kbps is a limited bit rate to send new blocks + txns + repetition blocks. Since new blocks have higher priority, whenever there is one, the other streams will wait until the new block finishes. So you won't be getting historic blocks all the time on the link. Again, the above RPCs are helpful to know what is being transmitted when, especially the `gettxqueueinfo` command.

VDelport, 2021-10-30:
I am now trying to transfer the UDP / IP data stream into my GNU radio project using netcat. This fails because netcat apparently changes the structure of the UDP packets. In any case, the recipient's bitcoind states:
UDP: Checksum error on message from 127.0.0.1:0

From previous tests I know that the transmission with netcat works if I forward the UDP packets in the transport stream via netcat and then unpack the UDP packets with tsp and pass them on to bitcoind.

Therefore I have to pack the UDP packets of the blocksat transmitter into a transport stream beforehand.

Can you tell me what the exact command line is to embed the UDP packets in the transport stream using tsp?

blockstreamsatellite, 2021-11-01:
Hi @VDelport ,

We don't use `tsp` at our Tx side to encapsulate UDP/IP over MPE and MPEG TS (we use pro DVB-S2 transmitter hardware instead). So I'm afraid I can't help you with the command line. However, I believe the `tsp` plugin you are looking for is `mpeinject`, followed by a plugin to encapsulate the MPE stream into MPEG TS (not sure which).

However, it is worth noting that:

1. You probably don't need `netcat` at all. Note you can configure `bitcoind` to send the UDP/IP stream directly to your GNU Radio transmitter. Why use `netcat` in the middle? You can use GNU Radio's UDP Source block to listen to the `bitcoind` traffic on port 4434.
2. If you send UDP/IP stream directly, without the MPE and MPEG TS layers, the Rx side should correspondingly assume the incoming traffic comes directly on IP packets. Are you trying to receive with the regular Blockstream Satellite receiver processing pipeline? That would not work since the Blockstream Satellite receiver assumes the `MPEG TS > MPE > IP > UDP` encapsulation, and I assume you would have only `Link Layer > IP > UDP` (not sure which link layer you are using). In other words, you wouldn't use `tsp` on the Rx side if you transmit the UDP/IP stream directly. Instead, you would feed the radio receiver's output directly into the `bitcoind` instance configured with the appropriate `udpmulticast` option.

VDelport, 2021-11-01:
>You can use GNU Radio's UDP Source block to listen to the bitcoind traffic on port 4434.

Yes, that's right. I tested it with the GNU Radio Block UDP Source and it works in principle. However, there is still an issue with the UDP payload size. I have to define a payload size in the GNU Radio Block UDP Source. With Wireshark I found out that the blocksat transmitter generally transmits UDP segments with a payload size of 1184 bytes. However, this is not always the case, which means that UDP segments with a smaller payload size are sometimes transmitted. That then leads to the above mentioned error in bitcoind.

Why are small UDP segments sometimes transmitted, especially after the blocksat transmitter has started?

blockstreamsatellite, 2021-11-01:
Is the payload size parameter on the UDP Source block an upper limit or a requirement? That is, should all UDP packets have the same size in order for the UDP Source block to work?

Bitcoin Satellite only sends small (variable-length) packets for the mempool transactions. All other packets are of fixed size, with a UDP payload of 1184 bytes, as you correctly observed. If you need to disable the variable-length packets, you can disable the transmission of mempool transactions by setting the following on your `tx.conf` file:

```
txn_per_sec=0
```

The reason why the txns are sent with variable length is because it would be very wasteful to zero-pad them to 1184 bytes, as they are typically shorter than 1184 bytes. In contrast, blocks are almost always larger than 1184 bytes, so they are FEC-encoded and padded.

VDelport, 2021-11-02:
>If you need to disable the variable-length packets, you can disable the transmission of mempool transactions by setting the following on your tx.conf file:
txn_per_sec=0

It works fine now. 

Thank you for your detailed and friendly support. :)

blockstreamsatellite, 2021-11-03:
You are welcome, @VDelport ! I'm glad everything is working now. :)

If you are running a satellite receiver continuously and would like to help us better plan and improve the worldwide coverage, please consider running your receiver with option `--report` on `blocksat-cli` to report your signal quality. More information is available at https://blockstream.github.io/satellite/doc/monitoring.html.

And please feel free to get in touch any time if you need more help.

Regards

closed 2021-11-03 by blockstreamsatellite