Blockstream/satellite #85: dvb_usb not detected on 20.04

3cl1ps, 2022-01-16:
Got the same issue on 5927: https://github.com/tbsdtv/linux_media/issues/209 
but couldn't make it work buy manually compiling drivers
so i installed ubuntu 18.04, now the dvb is detected.

blockstreamsatellite, 2022-01-24:
Hi @3cl1ps ,

Thanks for opening the issue. Which Linux distribution/version did you try before running on Ubuntu 18.04? Just so we can try to reproduce the problem and provide a better fix? Also, if possible, please share the result of `uname -a` from the system where you got the issue. That would be appreciated.

The CLI does have a workaround for the `ov9650.ko undefined` issue. However, it's currently applied to Fedora only, and for Linux kernel versions before 5.8.

Let us know if you need any help.

Thanks

3cl1ps, 2022-01-24:
it was on 20.04 ubuntu. 
ubuntu-20.04.3-preinstalled-server-arm64+raspi.img.xz
with latest kernel and patch. can't do an uname anymore.

blockstreamsatellite, 2022-01-24:
Thanks, @3cl1ps . Which Raspberry Pi model/version are you using?

3cl1ps, 2022-01-25:
rasp4, 4go.

closed 2023-03-03 by blockstreamsatellite