Blockstream/satellite #93: Has something changed in the configuration for receiving with the TBS receiver?

VDelport, 2023-01-09:
Hello,
I have successfully run Blockstream Satellite with TBS receiver in 2020. Now I wanted to put the system back into operation, but I didn't get any blocks or FEC chunks. Has something changed in the configuration?

My configurations:

config-channel.conf:
blocksat-ch]
DELIVERY_SYSTEM = DVBS2
FREQUENCY = 11484300
POLARIZATION = VERTICAL
SYMBOL_RATE = 1000000
INVERSION = AUTO
MODULATION = QPSK
VIDEO_PID = 32+33

bitcoin.conf:
debug=udpnet
debug=udpmulticast
udpmulticastloginterval=10
udpmulticast=dvb0_0,239.0.0.2:4434,172.16.235.25,1,blocksat-tbs-lowspeed
#udpmulticast=dvb0_1,239.0.0.2:4434,172.16.235.25,1,blocksat-tbs-highspeed

blocksat-dvb0_0.yaml:
networks:
   versions: 2
   renderer: networkd
   ethernet:
     dvb0_0:
       dhcp4: no
       optional: true
       addresses: [172.16.235.8/29]

Output after sudo blocksat-cli usb launch

------------------------------- Find DVB Adapter -------------------------------
FEHLER    FE_SET_VOLTAGE: Vorgang nicht zulässig
Found DVB-S2 adapter: TurboSight TBS 5927

----------------------------- Tuning DVB Receiver -----------------------------
Running dvbv5-zap
Verwendet wird LNBf UNIVERSAL
	universell, Europa
	Frequenzen: 10800 bis 11800 MHz, LO: 9750 MHz
	Frequenzen: 11600 bis 12700 MHz, LO: 10600 MHz
Demultiplexer »dvb0.demux0« wird verwendet.
Kanäle werden aus Datei »/home/blocksatdvbs2/.blocksat/config-channel.conf« gelesen.
Gerät TurboSight TBS 5927 DVB-S/S2 (/dev/dvb/adapter0/frontend0) Fähigkeiten:
     CAN_2G_MODULATION
     CAN_FEC_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_QPSK
DVB-API-Version 5.11, derzeitiges V5-Liefersystem: DVBS2
unterstützte Übertragungssysteme: 
     DVBS
    [DVBS2]
     DSS
Angenommen, Sie sind in DE

Einstellen auf 11484300 Hz
frequency: 11484,30 MHz, high_band: 0
SEC: setzt Spannung auf 13V
DiSEqC-KLANG: OFF
DiSEqC-KLANG: OFF
L-Band frequency: 1734,30 MHz (offset = 9750,00 MHz)
Video-PID 32
  dvb_set_pesfilter 32
RF     (0x01) Signal= -50,00dBm

Output after: ~$ bitcoind -connect=0 -debug=fec
2023-01-09T12:45:28Z Bitcoin Core version v22.0-satellite0.2.3 (release build)
2023-01-09T12:45:28Z Bitcoin Satellite version v0.2.3 (release build)
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -connect set -> setting -dnsseed=0
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -connect set -> setting -listen=0
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -listen=0 -> setting -upnp=0
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -listen=0 -> setting -natpmp=0
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -listen=0 -> setting -discover=0
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -listen=0 -> setting -listenonion=0
2023-01-09T12:45:28Z InitParameterInteraction: parameter interaction: -listen=0 -> setting -i2pacceptincoming=0
2023-01-09T12:45:28Z Assuming ancestors of block 00000000000000000008a89e854d57e5667df88f1cdef6fde2fbca1de5b639ad have valid signatures.
2023-01-09T12:45:28Z Setting nMinimumChainWork=00000000000000000000000000000000000000001fa4663bbbe19f82de910280
2023-01-09T12:45:28Z Using the 'shani(1way,2way)' SHA256 implementation
2023-01-09T12:45:28Z Using RdSeed as additional entropy source
2023-01-09T12:45:28Z Using RdRand as an additional entropy source
2023-01-09T12:45:28Z No wallet support compiled in!
2023-01-09T12:45:28Z Default data directory /home/blocksatdvbs2/.bitcoin
2023-01-09T12:45:28Z Using data directory /home/blocksatdvbs2/.bitcoin
2023-01-09T12:45:28Z Config file: /home/blocksatdvbs2/.bitcoin/bitcoin.conf
2023-01-09T12:45:28Z Config file arg: debug="udpnet"
2023-01-09T12:45:28Z Config file arg: debug="udpmulticast"
2023-01-09T12:45:28Z Config file arg: udpmulticast="dvb0_0,239.0.0.2:4434,172.16.235.25,1,blocksat-tbs-lowspeed"
2023-01-09T12:45:28Z Config file arg: udpmulticastloginterval="10"
2023-01-09T12:45:28Z Command-line arg: connect="0"
2023-01-09T12:45:28Z Command-line arg: debug="fec"
2023-01-09T12:45:28Z Using at most 125 automatic connections (1024 file descriptors available)
2023-01-09T12:45:28Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2023-01-09T12:45:28Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2023-01-09T12:45:28Z Script verification uses 3 additional threads
2023-01-09T12:45:28Z scheduler thread start
2023-01-09T12:45:28Z HTTP: creating work queue of depth 16
2023-01-09T12:45:28Z Using random cookie authentication.
2023-01-09T12:45:28Z Generated RPC authentication cookie /home/blocksatdvbs2/.bitcoin/.cookie
2023-01-09T12:45:28Z HTTP: starting 4 worker threads
2023-01-09T12:45:28Z init message: Loading banlist…
2023-01-09T12:45:28Z SetNetworkActive: true
2023-01-09T12:45:28Z Using /16 prefix for IP bucketing
2023-01-09T12:45:28Z Cache configuration:
2023-01-09T12:45:28Z * Using 2.0 MiB for block index database
2023-01-09T12:45:28Z * Using 8.0 MiB for chain state database
2023-01-09T12:45:28Z * Using 440.0 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2023-01-09T12:45:28Z init message: Loading block index…
2023-01-09T12:45:28Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2023-01-09T12:45:28Z Opening LevelDB in /home/blocksatdvbs2/.bitcoin/blocks/index
2023-01-09T12:45:28Z Opened LevelDB successfully
2023-01-09T12:45:28Z Using obfuscation key for /home/blocksatdvbs2/.bitcoin/blocks/index: 0000000000000000
2023-01-09T12:45:28Z LoadBlockIndexDB: last block file = 0
2023-01-09T12:45:28Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=1, size=293, heights=0...0, time=2009-01-03...2009-01-03)
2023-01-09T12:45:28Z Checking all blk files are present...
2023-01-09T12:45:28Z Opening LevelDB in /home/blocksatdvbs2/.bitcoin/chainstate
2023-01-09T12:45:28Z Opened LevelDB successfully
2023-01-09T12:45:28Z Using obfuscation key for /home/blocksatdvbs2/.bitcoin/chainstate: 2145d7baee056839
2023-01-09T12:45:28Z Loaded best chain: hashBestChain=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f height=0 date=2009-01-03T18:15:05Z progress=0.000000
2023-01-09T12:45:28Z init message: Verifying blocks…
2023-01-09T12:45:28Z  block index              47ms
2023-01-09T12:45:28Z Opening LevelDB in /home/blocksatdvbs2/.bitcoin/future_blocks
2023-01-09T12:45:28Z Opened LevelDB successfully
2023-01-09T12:45:28Z Using obfuscation key for /home/blocksatdvbs2/.bitcoin/future_blocks: 0000000000000000
2023-01-09T12:45:28Z block tree size = 1
2023-01-09T12:45:28Z nBestHeight = 0
2023-01-09T12:45:28Z init message: Loading P2P addresses…
2023-01-09T12:45:28Z Loaded 0 addresses from peers.dat  1ms
2023-01-09T12:45:28Z init message: Starting network threads…
2023-01-09T12:45:28Z DNS seeding disabled
2023-01-09T12:45:28Z net thread start
2023-01-09T12:45:28Z msghand thread start
2023-01-09T12:45:28Z loadblk thread start
2023-01-09T12:45:28Z Imported mempool transactions from disk: 0 succeeded, 0 failed, 0 expired, 0 already there, 0 waiting for initial broadcast
2023-01-09T12:45:28Z loadblk thread exit
2023-01-09T12:45:28Z UDP: multicast rx -  multiaddr: 239.0.0.2, interface: dvb0_0 (172.16.235.8), sourceaddr: 172.16.235.25, trusted: 1
2023-01-09T12:45:28Z UDP: Socket 26 bound to port 4434 for multicast group 0 blocksat-tbs-lowspeed
2023-01-09T12:45:28Z UDP: Initializing connection to 172.16.235.25:0...
2023-01-09T12:45:28Z udpprocess thread start
2023-01-09T12:45:28Z addcon thread start
2023-01-09T12:45:28Z init message: Done loading
2023-01-09T12:45:28Z udpwrite thread start
2023-01-09T12:45:28Z udpread thread start
2023-01-09T12:45:28Z UDP: Was unable to set write thread priority to SCHED_RR (permission denied)
2023-01-09T12:45:28Z UDP: Set write thread nice value to -1
2023-01-09T12:45:28Z udploadpartialblks thread start
2023-01-09T12:45:28Z Loading partial blocks from disk...
2023-01-09T12:45:28Z Loaded 0 partial blocks from disk
2023-01-09T12:45:28Z udploadpartialblks thread exit

VDelport, 2023-01-29:
sudo pip3 install blocksat-cli --upgrade failed:

Collecting blocksat-cli
  Downloading https://files.pythonhosted.org/packages/56/d6/e2c8ff2f1a66f068402e6d9fabd1a808a4a735950bffea166e9bb4f4dbff/blocksat_cli-0.4.4-py3-none-any.whl (176kB)
    100% |████████████████████████████████| 184kB 2.0MB/s 
Collecting zfec>=1.5.4 (from blocksat-cli)
  Downloading https://files.pythonhosted.org/packages/06/93/1afdfb33e5515845a62d946ca9d53b18477f77cd7608ab7563c4ebbd2dba/zfec-1.5.7.2.tar.gz (78kB)
    100% |████████████████████████████████| 81kB 2.2MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-irtk8bqa/zfec/

VDelport, 2023-01-29:
Ok, for that I have to install setuptool first, with
sudo apt-get install python3-setuptools

But then leads
sudo pip3 install blocksat-cli --upgrade
to a compiler error:

  zfec/_fecmodule.c:5:10: fatal error: Python.h: File or directory not found
      #include <Python.h>
               ^~~~~~~~~~
     compilation terminated.
     error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

blockstreamsatellite, 2023-01-30:
Hi @VDelport ,

Could you check which Python version you are running? By running the following:

```
python3 --version
```

If it is Python 3.11, the binary (wheel) of the `zfec` dependency (installed with blocksat-cli) won't be available for download, so `pip` will try to build it instead. For that, you will need to install `gcc` and the Python3 headers:

```
sudo apt install gcc python3-dev
```

After that, try the upgrade again:

```
sudo pip3 install blocksat-cli --upgrade
```

VDelport, 2023-02-03:
python3 --version
Python 3.6.9

VDelport, 2023-02-03:
Thank you, it works now.
I had to run sudo apt install python3-dev.
With
udpmulticast=dvb0_0,239.0.0.2:4434,172.16.235.25,1,blocksat-tbs
the average birate is about 1 Mbit/sec.
bitcoind output:
UDP multicast group 0: Average bit rate 1.08 Mbit/sec (blocksat-tbs)

Do the following configuration variants still exist?
blocksat-tbs-lowspeed
blocksat-tbs-highspeed

blockstreamsatellite, 2023-02-03:
Hi @VDelport ,

Good to know that it works now.

The `blocksat-tbs-lowspeed` and `blocksat-tbs-highspeed` were removed back in [March 2021](https://github.com/Blockstream/satellite/wiki/Satellite-Network-Update-on-March-19%2C-2021). You only need one `udpmulticast` line bound to `dvb0_0` on your bitcoin.conf file, which is what you have. The bitrate you are receiving looks correct.

If you would like to help us better monitor the coverage and service, please consider our [receiver status reporting feature](https://blockstream.github.io/satellite/doc/monitoring.html) 😄 .

We will close the issue for now. Please feel free to reopen it if you need more help.

Regards

closed 2023-02-03 by blockstreamsatellite