Blockstream/satellite #104: Having trouble getting a lock on the Europe transponder

feyris-tan, 2024-04-02:
Hello.

I'm thinking about setting up a node, but before I Install the software, I wanted to see If my antenna setup is capable of it. However, I'm having issues getting a lock on the Transponder.
I'm in the middle of Germany. According to https://satbeams.com/footprints?beam=5928 a 50 cm should be sufficient to get a good reception of Telstar 11N.

Using an Inverto SatPal, I've aligned my dish to 37.6W and tried to lock onto the transponder using CrazyScan and tsp, but neither programs were able to get a signal on 11505.4V using a TBS5972.

I don't think my dish alignment is off, because CrazyScan manages to get a lock on Telstar 11N's other transponders, just not 11505.4V. (checked using a list found on https://www.flysat.com/en/satellite/telstar-11n ) The IQ Graph on 11505.4V looks like QPSK however.

Here's what I tried so far:
- Tried different dishes (an 120cm Octagon branded one, and one 85cm fuba branded one)
- Tried different LNBs (one Bullseye BE01, one MTI AP8-T2B)
- Tried different tuners (Two TBS5927, one TBS5580, one TBS6903x PCIe)
- Tried different cabling towards the dish
- Judging from the source code, dvbv5-zap is used to tune to the transponder by the software, so I tried that manually, and it didn't get a lock either.
- I remember it working fine when I was DX'ing it in Automn 2023. Has something changed since back then?

Is there something else I should try?

blockstreamsatellite, 2024-04-02:
Hi @feyris-tan ,

First of all, thanks for your interest in the project. Unfortunately, we are currently experiencing a problem on the T11N and T18V carriers due to faulty parts at our teleports. We are waiting for delayed replacement parts to fix the problem. We apologize for the inconvenience. Once they are fixed, the service should be restored immediately.

Regarding antenna sizes, we recommend checking our antenna recommendation from our [network coverage map](https://blockstream.com/satellite/#satellite_network-coverage). Once you select a location, click "More Information" on the bottom right, which redirects to the [link analyzer](https://satellite.blockstream.space/link-analyzer/) page. You can find more detailed information about antennas and expected carrier-to-noise ratios (CNRs) there. 

In any case, the components you have used (like the TBS5927 and dishes) are compatible, so your setup should work when the service is restored.

Thanks for your patience.

CryptoIceMLH, 2024-07-04:
> Hi @feyris-tan ,
> 
> First of all, thanks for your interest in the service. Unfortunately, we are currently experiencing a problem on the T11N and T18V carriers due to faulty parts at our teleports. We are waiting for delayed replacement parts to fix the problem. We apologize for the inconvenience. Once they are fixed, the service should be restored immediately.
> 
> Regarding antenna sizes, we recommend checking our antenna recommendation from our [network coverage map](https://blockstream.com/satellite/#satellite_network-coverage). Once you select a location, click "More Information" on the bottom right, which redirects to the [link analyzer](https://satellite.blockstream.space/link-analyzer/) page. You can find more detailed information about antennas and expected carrier-to-noise ratios (CNRs) there.
> 
> In any case, the components you have used (like the TBS5927 and dishes) are compatible, so your setup should work when the service is restored.
> 
> Thanks for your patience.

Can we have an update on this please!

TechMindsYT, 2024-07-06:
I've spent two days trying to align the dish, to get a signal. and then I come across this.. Can we confirm if 11N is working yet? I'm using SDR. There appears to be a signal on 11505.4, but I just cannot get a lock. Bullseye LNB with 90cm dish.

1. Can we have an update please as to what we should expect to see right now.
2. I'm assuming we do use Vertical polarization for the LNB?

blockstreamsatellite, 2025-01-24:
As mentioned in https://github.com/Blockstream/satellite/issues/105#issuecomment-2613129523, the Telstar 11N service has resumed regular operation since Dec 31, 2024.

We'll close this ticket for now. Please feel free to reopen if you have any questions.

Regards

closed 2025-01-24 by blockstreamsatellite