Blockstream/satellite #114: Upgrade of blocksat-cli not possible

VDelport, 2025-01-23:
Hello,

trying to upgrade blocksat-cli with the command

sudo pip3 install blocksat-cli --upgrade

 fails.

Why is that? The operating system is Linux Mint 19 Cinnamon.

Best Regards

blockstreamsatellite, 2025-01-24:
Hi @VDelport ,

Thanks for your question. Could you share the error message you are seeing?

While Linux Mint is not a supported distribution, it may be possible to install the CLI normally, provided the dependencies are installed. 

Could you check whether the installed Python version is at least 3.8 or later? To do that, run:

```
python3 --version
```

Python 3.8 and older versions [have reached end-of-life](https://devguide.python.org/versions/) status. Hence, currently, the Blocksat CLI supports Python 3.9 to 3.13.

Regards

VDelport, 2025-01-27:
python3 --version
Python 3.6.9

blockstreamsatellite, 2025-02-04:
Thanks for confirming. So, this is indeed a Python version that is no longer supported. Please try running on a more recent Python version and, ideally, a supported Linux distribution (Ubuntu LTS, Fedora, or Debian).

VDelport, 2025-02-06:
Ok, if Ubuntu LTS is recommended, why doesn't it work anyway (see Issue #115)?