Blockstream/satellite #115: Installation of the driver for the TBS 5927 fails

VDelport, 2025-01-23:
Hello,
I tried to install the driver for the TBS 5927 receiver using the following command:

sudo blocksat-cli deps tbs-drivers

I get the following error message:

/home/delport/.blocksat/src/tbsdriver/media_build/v4l/ccs-core.c: In function 'ccs_set_ctrl':
/home/delport/.blocksat/src/tbsdriver/media_build/v4l/ccs-core.c:668:21: error: too many arguments to function 'pm_runtime_get_if_active'
  668 |         pm_status = pm_runtime_get_if_active(&client->dev, true);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/delport/.blocksat/src/tbsdriver/media_build/v4l/ccs-core.c:22:
./include/linux/pm_runtime.h:75:12: note: declared here
   75 | extern int pm_runtime_get_if_active(struct device *dev);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/delport/.blocksat/src/tbsdriver/media_build/v4l/cx25840-firmware.o
make[4]: *** [scripts/Makefile.build:243: /home/delport/.blocksat/src/tbsdriver/media_build/v4l/ccs-core.o] Fehler 1
make[4]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[3]: *** [/usr/src/linux-headers-6.8.0-51-generic/Makefile:1925: /home/delport/.blocksat/src/tbsdriver/media_build/v4l] Fehler 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-51-generic'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/home/delport/.blocksat/src/tbsdriver/media_build/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2
Traceback (most recent call last):
  File "/usr/local/bin/blocksat-cli", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/blocksatcli/main.py", line 85, in main
    args.func(args)
  File "/usr/local/lib/python3.10/dist-packages/blocksatcli/dependencies.py", line 748, in drivers
    runner.run(["make", nproc_arg], cwd=media_build_dir)
  File "/usr/local/lib/python3.10/dist-packages/blocksatcli/util.py", line 323, in run
    res = subprocess.run(cmd,
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make', '-j16']' returned non-zero exit status 2.

Best regards.

VDelport, 2025-01-27:
delport@ftkt-nb06: uname -a
Linux ftkt-nb06 6.8.0-51-generic #52~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Dec  9 15:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

delport@ftkt-nb06: cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
UBUNTU_CODENAME=jammy

delport@ftkt-nb06: lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy

VDelport, 2025-02-04:
Hello, is there any solution to the problem?

blockstreamsatellite, 2025-02-04:
Hello, @VDelport .

Yes, this is a known issue with this kernel version. The following command should fix it:

```
blocksat-cli deps tbs-drivers --backport-disable v6.8-ccs.patch
```

VDelport, 2025-02-05:
After installing the driver I get the following error message:

blocksat-gui
2025-02-05 16:32:25 WARNING  Could not find the dvbnet interface name. Is the TBS 5927 receiver running?
2025-02-05 16:32:25 WARNING  Assuming interface name dvb0_0.
dmesg: Lesen des Kernelpuffers ist fehlgeschlagen: Vorgang nicht zulässig
2025-02-05 16:32:26 ERROR    No DVB-S2 adapters found
2025-02-05 16:32:27 INFO     D-Bus configuration file is missing.
2025-02-05 16:32:27 INFO     Polkit action file is missing.

blockstreamsatellite, 2025-02-05:
This seems to be a permission issue. Try adding your user to the `video` group with the following command:

```
sudo usermod -a -G video $LOGNAME
```

VDelport, 2025-02-06:
sudo usermod -a -G video $LOGNAME
does not lead to a solution:

blocksat-gui
2025-02-06 10:39:39 WARNING Could not find the dvbnet interface name. Is the TBS 5927 receiver running?

2025-02-06 10:39:39 WARNING Assuming interface name dvb0_0.
dmesg: Reading kernel buffer failed: operation not permitted
2025-02-06 10:39:40 ERROR No DVB-S2 adapters found

Obviously no interface dvb0_0 is created. What commands can be used to create this interface without having to use blocksat-cli?

blockstreamsatellite, 2025-02-06:
First, please note that the `usermod -a -G video` command does not take effect immediately in the terminal session. You would need to log out and back in for it to take effect. Alternatively, you could run the following command to apply the group in the current session:

```
newgrp video
```

Second, there is a chance your kernel got updated without you noticing, in which case you would have lost the TBS drivers and would need to reinstall them. Run `uname -r` again. Do you still have kernel `6.8.0-51-generic`? Maybe it got updated to `6.8.0-52-generic` while rebooting. See if that is the case.

If you lost the TBS drivers, the following command would not return anything:

```
lsmod | grep dvb
```

On the other hand, if the drivers are still there, and you have the TBS receiver connected, you would see something like:

```
dvb_usb_tbs5520se      20480  0
dvb_usb                45056  1 dvb_usb_tbs5520se
dvb_core              184320  1 dvb_usb
rc_core                73728  1 dvb_usb
videobuf2_vmalloc      20480  1 dvb_core
videobuf2_common       90112  3 videobuf2_vmalloc,dvb_core,videobuf2_memops
mc                     81920  3 si2157,dvb_core,videobuf2_common
```

> The above is for the TBS5520SE but should be similar for the TBS5927.

If that is all confirmed, go ahead and reinstall the drivers:

```
blocksat-cli deps tbs-drivers --backport-disable v6.8-ccs.patch
```

Once the drivers are reinstalled, you can check the `lsmod | grep dvb` command above or run `blocksat-cli usb list` to see the device listed. Alternatively, you can check the `dmesg` logs with the following command:

```
sudo dmesg | grep dvb
```

If you would like to prevent this issue from happening again (kernel updating on reboot), you can configure it to be held with a command like the following:

```
sudo apt-mark hold linux-image-$(uname -r)
```

Lastly, for your question:

> What commands can be used to create this interface without having to use blocksat-cli?

You can run the USB configuration in dry-run mode to see the commands that are executed:

```
blocksat-cli usb cfg --dry
```

VDelport, 2025-02-10:
>Maybe it got updated to 6.8.0-52-generic while rebooting. See if that is the case.
Yes, it is the case.

lsmod | grep dvb returns nothing.

And now?

VDelport, 2025-02-10:
Hello,
I have now found the solution to the problem. 
I had to disable secure boot.