Blockstream/satellite #116: blockstream-satellite cannot be found

VDelport, 2025-05-28:
I'm trying to install blockstream satellite on a Debian GNU/Linux 12 (bookworm) using the following commands, according to your instructions:

sudo add-apt-repository https://aptly.blockstream.com/satellite/debian/

sudo apt-key adv --keyserver keyserver.ubuntu.com \
    --recv-keys 87D07253F69E4CD8629B0A21A94A007EC9D4458C
    
sudo apt-get update

sudo apt-get install blockstream-satellite

After the last command, I get the response that blockstream stallite is not found.

VDelport, 2025-06-03:
Hello, is there any solution to the problem?

blockstreamsatellite, 2025-06-03:
Hi @VDelport ,

Thanks for reporting the issue. We have identified that the package was indeed missing in our Debian repository. 

The problem was fixed. It should be available now. Could you try again?

Please note that you may need to run the `add-apt-repository` part twice. Alternatively, you could follow the following slightly more manual sequence of commands:

```bash
echo "deb https://aptly.blockstream.com/satellite/debian/ bookworm main" | sudo tee /etc/apt/sources.list.d/blockstream.list
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 87D07253F69E4CD8629B0A21A94A007EC9D4458C
sudo apt-get update
sudo apt-get install blockstream-satellite
```

The latter adds the apt source directly to a file at `/etc/apt/sources.list.d/` instead of using the `add-apt-repository` command. However, note that when doing so, you must use the correct Debian release codename (e.g., `bookworm`) on the line. You can check yours by running:

```
lsb_release -cs
```

Let us know if that helps.